bulletphysics/bullet3 #1094: Deleting void * is undefined behavior.

iSLC, 2017-05-05:
In `examples/MultiThreading/b3Win32ThreadSupport.cpp` method `void b3Win32ThreadSupport::stopThreads()` an attempt to delete a void * is made with `delete threadStatus.m_lsMemory;`

I see two approaches to fixing this. First approach, introduce another function to delete the memory indirectly. The second one, add an extra argument to `b3Win32lsMemorySetupFunc`, such that if the argument is null, the memory is allocated and returned otherwise the given memory pointer is released. I'm not very fond of the second one but that's one way of dealing with it.

erwincoumans, 2017-05-06:
Indeed, next to a b3Win32lsMemorySetupFunc/b3PosixlsMemorySetupFunc we need a b3Win32lsMemoryDestroyFunc/b3PosixlsMemoryDestroyFunc.

erwincoumans, 2018-12-23:
This is fixed now. We use
~~~~~~~~~
  if (threadStatus.m_lsMemoryReleaseFunc)
                {
                        threadStatus.m_lsMemoryReleaseFunc(threadStatus.m_lsMemory);
                }
~~~~~~~~~

closed 2018-12-23 by erwincoumans