bulletphysics/bullet3 #1100: Virtual camera

tjacobs, 2017-05-06:
Will Bullet Physics have a virtual camera like Gazebo has? That'd be great to have for robotics sim, to output to a numpy 2d array. Like: http://gazebosim.org/tutorials?tut=ros_gzplugins#Pluginsavailableingazebo_plugins

Then I can run TensorFlow models on the vision.

erwincoumans, 2017-05-06:
It already has, people use pybullet.getCameraImage for computer vision. It has a built-in CPU renderer, so it is cloud-friendly. Plenty of people use it with TensorFlow, and there are examples how to convert it into numpy arrays. It works with out without some numpy optimization, see 
bullet/examples/pybullet/examples/testrender.py and testrender_np.py
This is all documented in the [pybullet quickstart guide.
](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#)

closed 2017-05-06 by erwincoumans

tjacobs, 2017-05-07:
Oh thanks! Sorry I didn't see that. Will check it out!