bulletphysics/bullet3 #1104: Graphical mesh lag behind collision shape in debug draw.

manmohanbishnoi, 2017-05-08:
Using `m_RigidBody->getMotionState()->getWorldTransform(trans);` causes the graphical mesh to lag behind the collision shape in DebugDraw call.
While `btTransform trans = m_RigidBody->getWorldTransform();` makes the collision shape and graphical mesh to align perfectly.

*note: **m_RigidBody** is an instance of **btRigidBody**.

Steps to reproduce:

1. Create 2 box shapes. One for ground and another a small box (mass = 10Kg)
2. Set ground box to (0, 0, 0) and smaller box at (0, 600, 0)
3. enable wireframe debug drawing.

Kindly find the images for my test on this page:
http://www.bulletphysics.org/Bullet/phpBB3/viewtopic.php?f=9&t=11689

erwincoumans, 2017-05-08:
You are calling debugdraw before stepSimulation. Move it after stepSimulation.

closed 2017-05-08 by erwincoumans

manmohanbishnoi, 2017-05-08:
Tried that before posting the issue.
Still the graphical mesh lags behind the collision shape.

erwincoumans, 2017-05-08:
We are not going to debug your code.There is no magic, the synchronization of motion states happens in btDiscreteDynamicsWorld::stepSimulation. Make sure rendering and debug drawing BOTH happen right after stepSimulation.

benelot, 2017-05-08:
Maybe it is better to ask this question at the Bullet Physics forum:

[http://bulletphysics.org/Bullet/phpBB3/](http://bulletphysics.org/Bullet/phpBB3/)

manmohanbishnoi, 2017-05-09:
@erwincoumans sir,

I'm not asking you to debug my code.
I've checked several times with debugger and found that values returned by `btRigidBody->getWorldTransfrom()` and `btRigidBody->getMotionState()->getWorldTransfrom()` are different.

As a proof I did modify a few lines in **OpenGLGuiHelper.cpp** to prove my observation. Here is the [screenshot](http://i.imgur.com/wXXXdwP.png) of my observation.

Kindly look at the **Watch1** panel at the bottom of image and notice the difference in m_origin values.

 ```
m_DynamicsWorld->stepSimulation(1 / 60.0f, 10);
btTransform trans, trans2;
trans2 = m_RigidBodies.at("CubeMesh")->m_Body->getWorldTransform();

m_RigidBodies.at("CubeMesh")->m_Body->getMotionState()->getWorldTransform(trans);
/*
 * using
 * m_RigidBodies.at("CubeMesh")->m_Body->getMotionState()->getWorldTransform(trans);
 *
 * causes the graphical mesh to lag behind the collision shape in DebugDraw call
 * Checking with debugger shows that the lag error keeps on increasing
 * */
// Added a Breakpoint here and found that btTransform.m_origin has different values for "trans" and "trans2"
glm::mat4 M;
trans.getOpenGLMatrix(glm::value_ptr(M));
m_RigidBodies.at("CubeMesh")->GetGraphicalObject()->Model = M;

if (m_DebugDraw) {
    m_DynamicsWorld->debugDrawWorld();
}
```

Looking at the demo code provided in bullet physics's source for [OpenGLGuiHelper.cpp](https://github.com/bulletphysics/bullet3/blob/master/examples/ExampleBrowser/OpenGLGuiHelper.cpp#L307) shows that **btCollisionObject->getWorldTransforms()** is used to sync Physics world to Graphics world instead of MotionState.

I am using btDefaultMotionState while adding rigid bodies to the dynamic physics world. If btMotionState works as it should then both Transforms should have same value for `m_origin` for a rigid body.

manmohanbishnoi, 2017-05-10:
BulletPhysics wiki page for [MotionState](http://www.bulletphysics.org/mediawiki-1.5.8/index.php/MotionStates#Interpolation) says

>  **Interpolation**
> Bullet knows how to interpolate body movement for you. As mentioned, implementation of interpolation is handled through MotionStates.
 > If you attempt to ask a body for its position, it will return the position at the end of the last physics tick. That's useful for many things, but for rendering you will want some interpolation. Bullet interpolates the transform of the body before passing the value to setWorldTransform.
 > If you want the non-interpolated position of a body [which will be the position as it was calculated at the end of the last physics tick], use btRigidBody::getWorldTransform() and query the body directly.
 > To get the interpolated values for rendering it is necessary to call StepSimulation(deltaTime) in your rendering loop. Only include a deltaTime parameter, no other arguments. 

Seems like while debug drawing the physics world, BulletPhysics uses non-interpolated position of a body while the graphical mesh uses interpolated position. This difference causes lag while debug drawing.

So the question is if a moving body has graphical mesh at _(0.0, 20.059, 0.0)_ and the debug drawer shows the collision shape at _(0.0, 20.017, 0.0)_; which one is the correct/actual position of rigid body?