bulletphysics/bullet3 #1106: DUSE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD compiler error

ahundt, 2017-05-09:
I'm getting a compiler error when I define `DUSE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD=ON` in my cmake build when trying to build commit  47eae8d, and the error is probably related to https://github.com/bulletphysics/bullet3/pull/850.

Full homebrew gist of installation process steps & output:
https://gist.github.com/ahundt/df791c901f89da9341f7760455c4bd9d

Relevant compiler errors:
```
cd /tmp/bullet-20170508-78186-o1zw4/build_cmake/examples/RobotSimulator && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++  -DBT_ENABLE_CLSOCKET -DBT_ENABLE_ENET -DBT_USE_DOUBLE_PRECISION -DHAS_SOCKLEN_T -DUSE_GRAPHICAL_BENCHMARK -DUSE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD -D_DARWIN -I/tmp/bullet-20170508-78186-o1zw4/src -I/tmp/bullet-20170508-78186-o1zw4/examples -I/tmp/bullet-20170508-78186-o1zw4/examples/ThirdPartyLibs -I/tmp/bullet-20170508-78186-o1zw4/examples/ThirdPartyLibs/enet/include -I/tmp/bullet-20170508-78186-o1zw4/examples/ThirdPartyLibs/clsocket/src  -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.11   -o CMakeFiles/App_RobotSimulator.dir/__/Importers/ImportMJCFDemo/BulletMJCFImporter.o -c /tmp/bullet-20170508-78186-o1zw4/examples/Importers/ImportMJCFDemo/BulletMJCFImporter.cpp
Undefined symbols for architecture x86_64:
  "btSoftBody::setTotalMass(double, bool)", referenced from:
      PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) in PhysicsServerCommandProcessor.o
  "btSoftBody::appendMaterial()", referenced from:
      PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) in PhysicsServerCommandProcessor.o
  "btSoftBody::randomizeConstraints()", referenced from:
      PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) in PhysicsServerCommandProcessor.o
  "btSoftBody::generateBendingConstraints(int, btSoftBody::Material*)", referenced from:
      PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) in PhysicsServerCommandProcessor.o
  "btSoftBody::scale(btVector3 const&)", referenced from:
      PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) in PhysicsServerCommandProcessor.o
  "btSoftBody::rotate(btQuaternion const&)", referenced from:
      PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) in PhysicsServerCommandProcessor.o
  "btSoftBody::translate(btVector3 const&)", referenced from:
      PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) in PhysicsServerCommandProcessor.o
[ 83%] Building CXX object examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory_GUI.dir/__/ExampleBrowser/OpenGLGuiHelper.o
  "btSoftBodyHelpers::CreateFromTriMesh(btSoftBodyWorldInfo&, double const*, int const*, int, bool)", referenced from:
      PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) in PhysicsServerCommandProcessor.o
cd /tmp/bullet-20170508-78186-o1zw4/build_cmake/examples/SharedMemory && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++  -DB3_USE_STANDALONE_EXAMPLE -DBT_USE_DOUBLE_PRECISION -DUSE_GRAPHICAL_BENCHMARK -DUSE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD -I/tmp/bullet-20170508-78186-o1zw4/src -I/tmp/bullet-20170508-78186-o1zw4/examples/ThirdPartyLibs -I/tmp/bullet-20170508-78186-o1zw4/examples/ThirdPartyLibs/Glew  -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.11   -o CMakeFiles/App_PhysicsServer_SharedMemory_GUI.dir/__/ExampleBrowser/OpenGLGuiHelper.o -c /tmp/bullet-20170508-78186-o1zw4/examples/ExampleBrowser/OpenGLGuiHelper.cpp
  "btSoftBodyHelpers::Draw(btSoftBody*, btIDebugDraw*, int)", referenced from:
      PhysicsServerCommandProcessor::renderScene() in PhysicsServerCommandProcessor.o
  "btSoftMultiBodyDynamicsWorld::addSoftBody(btSoftBody*, int, int)", referenced from:
      PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) in PhysicsServerCommandProcessor.o
  "btSoftMultiBodyDynamicsWorld::btSoftMultiBodyDynamicsWorld(btDispatcher*, btBroadphaseInterface*, btMultiBodyConstraintSolver*, btCollisionConfiguration*, btSoftBodySolver*)", referenced from:
      PhysicsServerCommandProcessor::createEmptyDynamicsWorld() in PhysicsServerCommandProcessor.o
  "btSoftBodyRigidBodyCollisionConfiguration::btSoftBodyRigidBodyCollisionConfiguration(btDefaultCollisionConstructionInfo const&)", referenced from:
      PhysicsServerCommandProcessor::createEmptyDynamicsWorld() in PhysicsServerCommandProcessor.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

erwincoumans, 2017-05-09:
Don't define DUSE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD. Does it work then? For pybullet use pip install pybullet. See also the pybullet quickstart guide at pybullet.org

ahundt, 2017-05-09:
#1107 is with DUSE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD disabled, but that one fails due to separate issues with the CMake configuration.

I use ROS and TensorFlow, which means using both the C & C++ APIs, and I was hoping to use a single installation for integrating both. Wouldn't that be best handled via the CMake install process?

ahundt, 2017-05-09:
I think this is likely solved fairly easily by adding whatever lib has a btSoftBodyRigidBodyCollisionConfiguration implementation to the target_link_libraries() call in whatever is compiling PhysicsServerCommandProcessor.

erwincoumans, 2017-05-09:
Sure, we should fix the cmake build. Why is pybullet not sufficient? pybullet works great with tensorflow. So you want a ROS - pybullet bridge? I think the c - api is not needed for that.

erwincoumans, 2017-05-09:
Which python version are you using? I think by default, the system version of Mac OSX is 32bit, but your cmake build seems to be 64bit.

ahundt, 2017-05-10:
I should be able to stick with python as long as I can manage performance bottlenecks, I'll explain why I asked. 

I work on a project called [costar](https://github.com/cpaxton/costar_stack) that uses more traditional approaches to planning and control with a combination of C++ and python code. For that reason, accessing a single scene instance from both python and C++ would be convenient but not a blocking requirement. A motivating example would be if bullet could take in high bandwidth data like either a point cloud from an RGBD sensor or a mesh continuously updated via C/C++. An example of such a mesh would be one created with data from SLAM running outside bullet. 

I'm planning to use something more traditional like the [Tasks robot model based control constrained optimization library](https://github.com/jrl-umi3218/Tasks) alongside machine learning for decision making. Actually since it looks like some of the IK stuff in bullet may be in progress Tasks may be a useful/interesting/relevant resource. The new pybullet functionality looks like it will be a great resource for the machine learning / reinforcement learning component, so thanks for the great work!

My cmake config was the same as #1107 I ran into this error first and it was with the addition of `-DUSE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD=ON`.

I actually encountered this compiler error both in Linux and OS X. #1107 is OS X only.

Here were the Linux flags and a [script to reproduce](https://github.com/ahundt/robotics_setup/blob/master/bullet.sh) (though soft body is removed in the link):

```

cmake .. \
-DCMAKE_C_FLAGS_RELEASE=-DNDEBUG \
-DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG \
-DCMAKE_INSTALL_PREFIX=/usr/local \
-DCMAKE_BUILD_TYPE=Release \
-DCMAKE_FIND_FRAMEWORK=LAST \
-DCMAKE_VERBOSE_MAKEFILE=ON \
-Wno-dev \
-DHAVE_CLOCK_GETTIME:INTERNAL=0 \
-DINSTALL_EXTRA_LIBS=ON \
-DBUILD_UNIT_TESTS=OFF \
-DUSE_DOUBLE_PRECISION=ON \
-DBUILD_BULLET2_DEMOS=ON \
-DBUILD_PYBULLET=ON \
-DBUILD_PYBULLET_NUMPY=ON \
-DBUILD_PYBULLET_CLSOCKET=ON \
-DBUILD_PYBULLET_ENET=ON \
-DBUILD_ENET=ON \
-DBUILD_CLSOCKET=ON \
-DBUILD_SHARED_LIBS=ON \
-DUSE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD=ON
```

PR #1132 (2017-05-21): btsoftbody fix #1106, compiler error in App_PhysicsServer_SharedMemory
BulletSoftBody was not linking to App_PhysicsServer_SharedMemory when 
cmake flag USE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD was ON.
fixes #1106
commit 3c4fd2ce (2017-05-21): Merge pull request #1132 from ahundt/btsoftbody

btsoftbody fix #1106, compiler error in App_PhysicsServer_SharedMemory
commit e719da34 (2017-05-21): btsoftbody fix #1106, compiler error in App_PhysicsServer_SharedMemory
BulletSoftBody was not linking to App_PhysicsServer_SharedMemory when USE_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD was enabled.
closed 2017-05-21 by erwincoumans