bulletphysics/bullet3 #1108: running examples after pip install pybullet

ahundt, 2017-05-09:
1. Can premade examples be run after `pip install pybullet`?

2. added to the introduction before the Hello pybullet world in the [pybullet quickstart guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3)? 
I'd add it myself but don't yet know the answer. :-)

3. Is there a way to run ExampleBrowser after `pip install pybullet` or is #830 still up to date and thus does ExampleBrowser still requires manual installation?

erwincoumans, 2017-05-09:
1) Yes, you can run most of the pybullet examples under Bullet/examples/pybullet/examples, after you installed using "pip install pybullet".
2) Isn't there an introduction before Hello pybullet?
3) The ExampleBrowser will automatically open when you use "import pybullet as p, p.connect(p.GUI)". The ExampleBrowser will run in 'PhysicsServer" mode.

closed 2017-05-09 by erwincoumans

ahundt, 2017-05-10:
@erwincoumans In 2 I meant to say: I think a one liner that runs ExampleBrowser would be a useful prelude to hello pybullet that lets users play with the functionality before coding. Sorry about the typo! 

Something like this, but this example exits immediately:

`python -c "import pybullet as p; p.connect(p.GUI); "`

Thanks for the help!

teristam, 2018-12-03:
The quickstart guide is super confusing right now. It is too long and reads like a documentation rather than a "quick" start. Some examples on running the ExampleBrowser is greatly appreciated.

ha7ilm, 2020-09-22:
Based on @ahundt 's command, this one-liner will not exit immediately:

    python3 -c "import pybullet as p; p.connect(p.GUI); input() "