bulletphysics/bullet3 #1109: Cannot load Universal Robot UR5 urdf

cpaxton, 2017-05-09:
Hoping to use this with the UR5. Unfortunately I'm having trouble getting the robot to load, and to appear in a GUI.

```
b3Printf: /home/chris.paxton/: cannot extract anything useful from mesh '../ur_description/meshes/ur5/visual/wrist2.dae'

b3Printf: b3Warning[examples/SharedMemory/TinyRendererVisualShapeConverter.cpp,412]:

b3Printf: issue extracting mesh from COLLADA/STL file ../ur_description/meshes/ur5/visual/wrist2.dae

b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,957]:

b3Printf: /home/chris.paxton/: cannot extract anything useful from mesh '../ur_description/meshes/ur5/visual/wrist3.dae'

b3Printf: b3Warning[examples/SharedMemory/TinyRendererVisualShapeConverter.cpp,412]:

b3Printf: issue extracting mesh from COLLADA/STL file ../ur_description/meshes/ur5/visual/wrist3.dae
```

cpaxton, 2017-05-09:
Correction: I can actually load the robot itself, but not the Collada DAE meshes.
[image]

These are from: https://github.com/ros-industrial/universal_robot

erwincoumans, 2017-05-09:
Collada DAE is an extremely complicated format and Bullet only supports a very simple subset of DAE. It would be best to convert it into Wavefront .obj or STL files. Do you have a pointer to the URDF+DAE files?

cpaxton, 2017-05-09:
Here's an example mesh: https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/meshes/ur5/visual/wrist1.dae

The collision meshes -- STLs -- all seem to load just fine.

cpaxton, 2017-05-09:
And here's an URDF loading these meshes:
[ur5_robot.urdf.txt](https://github.com/bulletphysics/bullet3/files/987764/ur5_robot.urdf.txt)

erwincoumans, 2017-05-09:
If I find some time, I can check if we can fix the DAE importer to accept those meshes. Otherwise, we could just use Blender to convert those DAE to .obj, that should work fine.

ahundt, 2017-05-18:
Is there a way to render joint locations easily such as with a triad? We are trying to add constraints for the gripper that goes with this arm and visualization would be helpful

erwincoumans, 2017-05-18:
Assuming you use pybullet, and pybullet.connect(pybullet.GUI), you can press 'w' for wireframe and press 'k' and 'l' (lower case L) to render joint frames/limits.
The COLLADA DAE file uses polylist, while Bullet only supports triangulated models (<triangles>). You may be able to convert/triangulate it using Blender, or convert it from DAE to .OBJ using Blender or other tool.

cpaxton, 2017-05-21:
Sounds good. OBJ conversion works for me.

closed 2017-05-21 by cpaxton