bulletphysics/bullet3 #1137: Something is wrong with rotational servo control

olegklimov, 2017-05-21:
For Humanoid knees, limits are (taken from .xml):

```python
>>> -160/180*np.pi
-2.792526803190927
>>> -2/180*np.pi
-0.03490658503988659
```

I'm using

```C++
b3JointControlSetDesiredPosition(..., ..., target_pos);
```
with `target_pos` the same -2.793, -0.035. 

Then I call function

```C++
b3GetJointState(..., ..., ..., &state);
```

after humanoid executes motion and settles in position, without gravity and self-collision, it returns in `state.m_jointPosition` values -2.376 and -0.035. Left limit is off. Leg doesn't reach its limit, I check this by changing motor control method to torque.

[image]

Furthermore, if I increase force, from 0.1 (tiny) to 40 (normal), bad things happen, knees can be turned  other direction, beyond limit.

[image]

None of this is visible when using torque control.

olegklimov, 2017-05-21:
(with torque 40 and more, knee doesn't go beyond limit.)

erwincoumans, 2017-05-21:
What happens if you make sure to set joint target positions are within limits (and not close to the limits)?

erwincoumans, 2017-05-21:
By the way, the rendering looks pretty bad without textures and shadows. You may want to use the build-in Bullet debug visualizer, it looks better.

olegklimov, 2017-05-21:
> What happens if you make sure to set joint target positions are within limits

For the left limit, the joint doesn't hit it. It stops 0.4 radians before hitting the limit. 

Generally, I didn't notice any jump in behavior when I set target within range or not.

erwincoumans, 2017-05-21:
So can you describe the bug more clearly, in case you set the joint target position motor safely within and not close to) limits?

olegklimov, 2017-05-21:
Both limits doesn't appear to be in place. Left limit never gets hit, and it bumps into right limit much sooner (by the same 0.4 rad).

olegklimov, 2017-05-21:
(I don't need it quickly, I just noticed it when debugging other things)

erwincoumans, 2017-05-21:
I created a quick reproduction case, and joint limits seem to work just fine, under joint motor position cotnrol. See this pybullet example, video will follow. Make sure to use the latest unmodified Bullet/pybullet version:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/humanoid_knee_position_control.py

Thanks!

erwincoumans, 2017-05-21:
Video is here: https://youtu.be/7vMKGUIc4S4

olegklimov, 2017-05-21:
Okay, it's my own fault. I've misinterpreted b3JointControlSetDesiredVelocity() for maximum joint speed. That's why it was always positive and skewed limits to the right. I've posted an edit for pybullet doc.

closed 2017-05-21 by olegklimov

erwincoumans, 2017-05-21:
Thanks for the report anyway, it motivated me to create a small example!