bulletphysics/bullet3 #1138: geometry, linear algebra, realtime fwd+inv kinematics & dynamics

ahundt, 2017-05-21:
**tl;dr** These are high level python/robot control API suggestions. Feel free to close if not applicable.

A [comment in issue 1136](https://github.com/bulletphysics/bullet3/issues/1136#issuecomment-302944843) mentioned plans for generic linear algebra tools for pybullet. I know bullet makes use of Featherstone, so the Tasks libraries may be good for a numpy style python API or perhaps just design ideas:

```python
import numpy as np
import eigen3 as e
import spacevecalg as sva
import rbdyn as rbd

# Featherstone Plucker transform, much like translation + quaternion pair

X_O_foot = sva.PTransformd(sva.RotZ(np.pi)*sva.RotX(-np.pi/2.), e.Vector3d(0.5, -0.3, 0.))
X_O_CoM = sva.PTransformd(e.Vector3d(-0.2, 0.8, 0.))

# Spherical joint & Quaternion

jSph = rbd.Joint(rbd.Joint.Spherical, True, 0, 'jSph')
jointResume(jSph)
q = e.Quaterniond(np.pi/4., e.Vector3d(1., 0., 0.5).normalized())
qParam = [q.w(), q.x(), q.y(), q.z()]
print 'rotation:'
print jSph.pose(qParam).rotation()
print 'angular motion:', jSph.motion([1., 0.2, 0.5]).angular().transpose()

# Creating Rigid Bodies

b0m = 2. # 2kg
b0c = e.Vector3d(0.5, 0.5, 0.) # center of mass at [0.5, 0.5, 0.] from the body origin
b0rI = e.Matrix3d.Identity() # rotational inertia at body origin
# here the second argument is the body first moment of mass (h = m c)
b0I = sva.RBInertia(b0m, b0m*b0c, b0rI)
# first body constructor that take sva.RBInertia as firt param
b0 = rbd.Body(b0I, 0, 'b0')

b1m = 4. # 4kg
b1c = e.Vector3d(0., 0.5, 0.) # center of mass at [0., 0.5, 0.] from the body origin
b1rI = e.Matrix3d.Identity() # rotational inertia at body origin
# here the second argument is the body center of mass (not the first moment of mass)
b1 = rbd.Body(b1m, b1c, b1rI, 1, 'b1')

# note that in the latest api only string ids like 'b2'
# are used, none of the manual numerical ids seen here
b2 = rbd.Body(1., e.Vector3d.Zero(), e.Matrix3d.Identity(), 2, 'b2')
b3 = rbd.Body(1., e.Vector3d.Zero(), e.Matrix3d.Identity(), 3, 'b3')
b4 = rbd.Body(1., e.Vector3d.Zero(), e.Matrix3d.Identity(), 4, 'b4')
b5 = rbd.Body(1., e.Vector3d.Zero(), e.Matrix3d.Identity(), 5, 'b5')

# Inverse Dynamics with Gravity Comp
# assume some multibody system is defined with mb, a list of joints, etc.
mbcID = rbd.MultiBodyConfig(mb)
mbcID.zero(mb)
mbcID.q = mbc.q

rbd.forwardKinematics(mb, mbcID)
rbd.forwardVelocity(mb, mbcID)
ID = rbd.InverseDynamics(mb)
ID.inverseDynamics(mb, mbcID)

jointTorque = map(list, mbcID.jointTorque)
print 'jointTorque:', jointTorque

# jointTorque: [[], [-24.525], [4.905000000000001], [8.047710038720992e-17], [8.495709211125343], [0.0, 0.0, 0.0]]

```

 **Clean tools to define and solve for goals & constraints for any part of a robot**, such as end effectors, which are then solved for via Quadratic Programming based approaches:
 * Articular position, speed and torque limits, Collision avoidance
 * Multi-robot contact (can solve problems involving multiple robots in contact)
 * Task Target Goals (Inverse kin/dyn):
    * Link Position/Orientation, Velocity, Center of Mass (CoM) target, Momentum, Contact force target

Unfortunately I don't have a python example on hand so here is a
[C++ example of Tasks using position and orientation goals](https://github.com/ahundt/grl/blob/master/include/grl/vrep/InverseKinematicsVrepPlugin.hpp#L616)
```C++
        tasks::qp::QPSolver solver;
        
        ////////////////////////////////////////////////////
        // Set position goal of the arm
        sva::PTransformd targetWorldTransform;
        // ...snip...

        // mbs defines the robot structure
        // mbcs defines where the robot is now
        tasks::qp::PositionTask posTask(rbd_mbs_, simulatedRobotIndex, ikGroupTipName_,targetWorldTransform.translation());
        tasks::qp::SetPointTask posTaskSp(rbd_mbs_, simulatedRobotIndex, &posTask, 50., 10.);
        tasks::qp::OrientationTask oriTask(rbd_mbs_,simulatedRobotIndex, ikGroupTipName_,targetWorldTransform.rotation());
        tasks::qp::SetPointTask oriTaskSp(rbd_mbs_, simulatedRobotIndex, &oriTask, 50., 1.);
        tasks::qp::PostureTask postureTask(rbd_mbs_,simulatedRobotIndex,rbd_preferred_mbcs_[simulatedRobotIndex].q,1,0.01);

        ////////////////////////////////////
        // Set joint limits

        // ...snip...

        tasks::qp::DamperJointLimitsConstr dampJointConstr(rbd_mbs_, simulatedRobotIndex, {lBound, uBound},{lVelBound, uVelBound}, 0.125, 0.025, 1., simulationTimeStep);

        // Test add*Constraint
        dampJointConstr.addToSolver(solver);
        BOOST_VERIFY(solver.nrBoundConstraints() == 1);
        BOOST_VERIFY(solver.nrConstraints() == 1);

        solver.nrVars(rbd_mbs_, {}, {});
        solver.updateConstrSize();

        solver.addTask(&posTaskSp);
        solver.addTask(&oriTaskSp);
        solver.addTask(&postureTask);

        // ...snip...
        solver.solve(rbd_mbs_, rbd_mbcs_);
        // ...snip...
```
[Video of Robot Following Simulated Real Time Constrained Optimization Motion Task in V-REP](https://youtu.be/nxYoOChgMKg)

- Tasks Library Stack from high level to low level APIs
    - [github.com/jrl-umi3218/sva_rbdyn_tutorials](https://github.com/jrl-umi3218/sva_rbdyn_tutorials) Python tutorials
    - [Tasks](https://github.com/jrl-umi3218/Tasks) Quadratic Programming based realtime arm control with position, velocity, contact forces, etc.
    - [RBDyn](https://github.com/jrl-umi3218/RBDyn) Featherstone Rigid Body Dynamics
    - [SpaceVecAlg](https://github.com/jrl-umi3218/SpaceVecAlg) Featherstone Spatial Vector Algebra
    - [shell script to build/install the whole stack](https://github.com/ahundt/robotics_setup/blob/master/robotics_tasks.sh)

erwincoumans, 2017-05-21:
Thanks for the pointers. I was hoping that users with use numpy and eigen as much as possible. In addition, the API is much simpler if we leave the creation in URDF/SDF/MJCF, since programmatically creating multi bodies clutters the API a lot. Now we already have pybullet APIs for inverse dynamics and inverse kinematics. It may be good to expose lower level parts of 'stepSimulation', such as forwardKinematics, forwardVelocity, like MuJoCo does. I wouldn't hold my breath for any of this, since there is a lot of other work that has higher priority.

Where do your code snippets come from?

ahundt, 2017-05-22:
Just occurred to me that rather than numpy, math/transform calculations that will done at every simulation step might be better done via a tf graph rather than a numpy style API. I know tf also uses eigen internally. I haven't tested it but the new function [session.make_callable()](https://www.tensorflow.org/versions/master/api_docs/python/tf/Session#make_callable) in [tf1.2.0](https://github.com/tensorflow/tensorflow/releases/tag/v1.2.0-rc0) is supposed to lower the overhead for frequent calls and could make tf style calculations a more feasible option.

erwincoumans, 2017-05-22:
Either way is fine, both should be possible. Have you tried running humanoid_running.py? This one uses Tensorflow for inference. It is reasonably easy to replace TF with numpy for that example, I think they did this for OpenAI Roboschool (which is based on Bullet C-API, but they added their own framework/simple renderer on top, but in my opinion pybullet and its renderer is more powerful than what Roboschool has at the moment).

ahundt, 2017-05-23:
looks cool, glad I have it running but I meant something a bit different. which could potentially make pybullet scene calculations and image generation 10x faster than if they were implemented the way numpy is. Tensorflow happens to be used for machine learning, but really it is just a math library that creates a graph of mathematical operations saved in a protobuf format that can be run much faster than the python interpreter. 

This alternative design is one where pybullet could supply tensorflow ops to retrieve and set transforms, load segmentation images, rendered images, do quaternion math, etc. In other words the images might never need to be copied from c to python.

Essentially, use the ability to [implement new ops](https://www.tensorflow.org/extend/adding_an_op) and add [custom data readers](https://www.tensorflow.org/extend/new_data_formats) to define functions under the namespace `pybullet.tf`.

example:
```python
import tensorflow as tf
# new pybullet tensorflow functionality
import pybullet.tf as pbtf 

# op that when run returns true if the simulation is running,
# false otherwise.
simulation_running_op = pbtf.SimulationIsRunning()

# this returns a tensorflow op, not a numpy array
camera_op = pbtf.getCameraImage(width,height)

# can have some configuration params for how it works...
random_camera_op = pbtf.randomizeCameraPose(camera_op)

# this is some user defined function 
# that loads some regular tf model
# and weights from disk, let's say it
# is predicting segmentation
model_tf_op = my_load_model(random_camera_op)

segmentation_op = pbtf.getCameraSegmentation(random_camera_op)

# some train function
training_op = my_train_fn(model_tf_op, segmentation_op)

with tf.Session() as sess:
  train_with_pybullet = tf.while_loop(training_op, simulation_running_op)
  # probably needs some time limit parameter or something
  sess.run(train_with_pybullet)  

```

Additional ops can exist for getting transforms between objects, creating and configuring qp solvers, basically any functionality the user wants, but they get a huge performance boost from it being tensorflow ops instead of directly running python code.

Hopefully my explanation made sense?

erwincoumans, 2017-05-23:
Yes. Actually I'm in the brain robotics team surrounded by the Tensorflow developers, it makes sense to write some custom ops at some stage :-)

erwincoumans, 2017-05-30:
Another small step: pybullet.multiplyTransforms and pybullet.invertTransform: https://github.com/bulletphysics/bullet3/pull/1158

Given a local transform (childLocalPos, childLocalOrn) and parent transform (parentWorldPos, parentWorldOrn), you can compute the child world transform using childWorldPos, childWorldOrn = pybullet.multiplyTransform(parentPos, parentOrn, childLocalPos, childLocalOrn)

ahundt, 2017-05-31:
**tl;dr**

Suggests a simplified multiplyTransform API, and [Cython](cython.org) to reduce complexity and improve the ease of use, also see #1163.

**Simplified multiplyTransform API**

#1158, which [goes to and from two arrays](https://github.com/bulletphysics/bullet3/pull/1158/files#diff-25c9b5bbc7f119ebe748c74a68ade10cR3184), seems to be more complex and higher overhead than needed. Here is a nice design concept that might fit this use case:

```python
import pbyullet as p
import numpy as np

# Transform can contain a translation + quaternion pair
t1 = p.Transform(p.RotZ(np.pi)*p.RotX(-np.pi/2.), p.Vector3(0.5, -0.3, 0.))

# handle objects
objectHandle = p.Handle(uniqueId=5, jointIndex=3)
relativeToObjectHanle = p.Handle(uniqueId=3, jointIndex=None)

# transform object
t2 = p.Transform(object, relativeToObject)

# simpler multiply operation
t3 = t1*t2
```

**Minimizing both Dependencies and Custom Code**

It seems the simple multiply operation in #1158 needed a [ton of PyObject interface code](https://github.com/bulletphysics/bullet3/pull/1158/files#diff-9190d83a6608c2072d75f4091c104b26R5672), which might become painful to maintain as it grows. [Cython](cython.org) was [developed in part at Google](https://research.google.com/pubs/pub36727.html) and may be another good way to generate the APIs and automate, reduce, or eliminate much of the PyObject C++ code. #1163 may be another approach to this issue.

erwincoumans, 2017-05-31:
I intentionally use manual Python bindings, no interest in Cython, swig or other frameworks.

I guess what you are asking for is a conversion from the pos,rot from Bullet to numpy or Tensorflow or another linear math library? I'm currently not planning to rewrite numpy/Tensorflow.

Or are you saying that there are no good linear algebra packages available?

ahundt, 2017-06-01:
> I intentionally use manual Python bindings, no interest in Cython, swig or other frameworks. I don't intend to grow the linear algebra parts much more, just a little bit :-)

Ah, I looked at Cython generated bindings and they didn't seem too bad, perhaps they are? The generated code looked like it used regular PyObject bindings, so it looked like Cython is only a dependency at install time.

> I guess what you are asking for is a conversion from the pos, rot from Bullet to numpy or Tensorflow or another linear math library? 

Essentially, yes. Naive conversion is very easy but also very very slow. I'm planning to move lots of transforms and images around, hopefully many times a second so that will matter. :-)

> I'm not planning to write a large linear math library.

I agree that would definitely be time consuming! Will the python API to the featherstone implementation and for configuring the IK solvers / optimizers be similarly limited?

I'm advocating bindings to an existing one via a backend, as I detail in https://github.com/bulletphysics/bullet3/issues/1163#issuecomment-305346913, with something reasonable provided by default. That's to avoid complexity while still supporting the necessary tools with fast performance.

erwincoumans, 2017-06-01:
Now we are talking. If you need to optimize for a specific case, like rendering many images and getting many transforms, there are a couple of optimizations.

There is already a numpy optimized version of getCameraImage, that is much faster than the non-numpy version (the slow version copies individual pixel components in python lists, the fast version just maps the array of memory to a numpy friendly format). 

For ray tests, there is a batch API, to perform multiple ray intersection tests.

Ultimately there will be some use-case that requires faster interoper between pybullet and Tensorflow or numpy, and if we think it will be generally beneficial we'll figure out some faster solution, possibly using some Tensorflow OP extensions, but I cannot promise that right now.

On my wishlist is a GPU physics backend (using the Bullet OpenCL physics implementation), that will allow large scale or many small parallel simultaneous simulations on GPU.