bulletphysics/bullet3 #1139: Segmentation Fault example browser when Inverse Dynamics URDF not found

kopp, 2017-05-21:
I have installed [bullet on Arch Linux](https://www.archlinux.org/packages/extra/i686/bullet/).  The example browser works fine for the first examples, but when I try to run *Inverse Dynamics URDF*, I get a segmentation fault.  Unfortunately, I cannot open the example browser again, because whenever it starts, it tries to load the most recent demo -- which creates the segfault.

Here is a backtrace generated by gdb:
```
% gdb bullet3_examplebrowser
Reading symbols from bullet3_examplebrowser...(no debugging symbols found)...done.

 gdb > run
Starting program: /usr/bin/bullet3_examplebrowser 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
X11 functions dynamically loaded using dlopen/dlsym OK!
glewXInit dynamically loaded using dlopen/dlsym OK
glewXInit OK
Creating context
Created GL 3.0 context
Direct GLX rendering context obtained
Making context current
glewOpenGL11Init dynamically loaded using dlopen/dlsym OK
glewOpenGL11Init OK!
GL_VENDOR=Intel Open Source Technology Center
GL_RENDERER=Mesa DRI Intel(R) Ivybridge Mobile 
GL_VERSION=4.2 (Core Profile) Mesa 17.1.0
GL_SHADING_LANGUAGE_VERSION=4.20
pthread_getconcurrency()=0
b3Printf: Selected demo: Inverse Dynamics URDF
URDF file not found

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff6a53f1b in btMultiBody::forwardKinematics(btAlignedObjectArray<btQuaternion>&, btAlignedObjectArray<btVector3>&) () from /usr/lib/libBulletDynamics.so.2.86

 gdb > bt
#0  0x00007ffff6a53f1b in btMultiBody::forwardKinematics(btAlignedObjectArray<btQuaternion>&, btAlignedObjectArray<btVector3>&) () from /usr/lib/libBulletDynamics.so.2.86
#1  0x00000000004bfcbf in InverseDynamicsExample::stepSimulation(float) ()
#2  0x00007ffff7a9bf0f in OpenGLExampleBrowser::update(float) () from /usr/lib/libBulletExampleBrowserLib.so.2.86
#3  0x0000000000448561 in main ()
```

It looks like the urdf file is simply not included in the package.
Is there a workaround, though, to re-start the example browser?  (I.e. can I make the browser forget what the last example was?)

erwincoumans, 2017-05-21:
The data folder should be distributed and placed/copied into the data folder, next to the location of the BulletExampleBrowser exacutable. We should fix the crash, but the example browser is useless without the data folder. Can you get in touch with the developer of Arch Linux and ask to include the data folder? If the data folder is too big, we can extract a useful set.

For now, I would recommend compiling Bullet from source code from http://github.com/bulletphysics/bullet3, then it should work fine. Finally, you can use ExampleBrowser 
--start_demo_name=CoordinateSystemDemo to start a different demo, or locate and delete the file called 0_Bullet3Demo.txt (likely in your current working directory). Thanks for the report!

kopp, 2017-05-23:
I opened [this bug report for the arch linux package](https://bugs.archlinux.org/task/54149?project=1&cat%5B0%5D=2&string=bullet) and linked this report there.

Thank you for the workaround, it ... well ... worked :-)

Maybe as a side-note: It would be great, if the example browser had an option such as `--help` to list options.

svenstaro, 2017-06-20:
Fixed in the package.

closed 2021-04-26 by erwincoumans