bulletphysics/bullet3 #1140: URDF joint mimic tag not supported

cpaxton, 2017-05-21:
The Bullet URDF parser does not appear to support the URDF mimic joint tags, as used in models such as the [Robotiq c-series gripper](https://github.com/ros-industrial/robotiq). Is there an alternative or preferred way to handle this in Bullet3?

erwincoumans, 2017-05-21:
There is no support yet for the mimic_joint, but we can add it. Previously we implemented a differential gear constraint (for btRigidBody/btTypedConstraint). We can implement the gear constraint for btMultiBody and use it to mimic the mimic joint ;-)

erwincoumans, 2017-05-21:
I just noticed Gazebo/SDF discussion about mimic: https://github.com/ros-simulation/gazebo_ros_pkgs/issues/251

and Drake about the mimic joint:
https://github.com/RobotLocomotion/drake/issues/2940

Both Gazebo and Drake seem to not added easy/automatic support for it yet.

cpaxton, 2017-05-21:
I was hoping to use addConstraint, actually, since that would be a more realistic depiction of the underlying physics for a single finger, but that's giving me some trouble. I tend to end up with a constantly rotating mess of fingertips like this, which I presume means I'm just setting up the constraints wrong:
[image]

As far as the mimic joint tag goes, there's a simple gazebo plugin with a decent implementation [here](https://github.com/StanleyInnovation/robotiq_85_gripper/blob/master/robotiq_85_simulation/roboticsgroup_gazebo_plugins/src/mimic_joint_plugin.cpp), which does something very similar to what you're suggesting.

Setting position commands for the whole set of fingertips works nicely enough for my purposes, though; thank you.

cpaxton, 2017-05-21:
Gazebo itself hasn't added mimic support, but for most purposes the hack works. Half the reason people still use gazebo is the existence of all these nice community plugins for different things!

erwincoumans, 2017-05-21:
You may also want to look at some of the Bullet examples in the ExampleBrowser. We have a few different solutions for a gripper, to make sure the joint angle of one finger is same as the other. One is based on some additional joint, and another one is based on a constraint motor, with velocity/position targets.

Since we are using Python with pybullet, it should be easier for people to contribute and share solutions. Some kind of plugin system might be good in the future indeed. And once the Bullet C-API support programmatic creation of multibodies (instead of loading from SDF/URDF/MJCF) we could upgrade Gazebo to use Bullet properly.

cpaxton, 2017-05-21:
Fantastic, thanks for pointing me to these. I'll close the issue since there are plenty of ways of implementing the functionality I want.

closed 2017-05-21 by cpaxton