bulletphysics/bullet3 #1141: humanoid_running falls down

ahundt, 2017-05-23:
By falls down I mean quits immediately, I just couldn't help including a pun:

```
± python humanoid_running.py
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce GT 750M OpenGL Engine
Version: 4.1 NVIDIA-10.10.14 310.42.25f02
GLSL: 4.10
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0
MotionThreadFunc thread started
Traceback (most recent call last):
  File "humanoid_running.py", line 14, in <module>
    plane, human  = p.loadMJCF("mjcf/humanoid_symmetric.xml",flags = p.URDF_USE_SELF_COLLISION_EXCLUDE_ALL_PARENTS)
AttributeError: 'module' object has no attribute 'URDF_USE_SELF_COLLISION_EXCLUDE_ALL_PARENTS'
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
destroy semaphore
semaphore destroyed
Thread with taskId 0 exiting
Thread TERMINATED
destroy main semaphore
main semaphore destroyed

± python humanoid_running_3.py
  File "humanoid_running_3.py", line 66
    pos, vel, *other = p.getJointState(human, j)
              ^
SyntaxError: invalid syntax

```

erwincoumans, 2017-05-23:
It looks like you are not using the very latest version of pybullet?

erwincoumans, 2017-05-23:
In addition, you should not run the _3 version, but humanoid_running.py That one doesn't have the *other thing that breaks on some python versions.

ahundt, 2017-05-23:
checked out master, pulled, ran:

```
sudo python setup.py install
```

And still saw a window appear and disappear in maybe 0.2 seconds. Maybe I should try on my 16.04 workstation.

erwincoumans, 2017-05-23:
Do you have Tensorflow installed and working? If not, please provide output. Perhaps you need to do 'uninstall' and make sure you have no other pybullet versions?

ahundt, 2017-05-23:
uninstalled and reinstalled, now he runs for a few seconds 58 meters, he is pretty quick. Maybe the demo should loop?

```
± python humanoid_running.py
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce GT 750M OpenGL Engine
Version: 4.1 NVIDIA-10.10.14 310.42.25f02
GLSL: 4.10
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0
MotionThreadFunc thread started
2017-05-22 21:08:30.215079: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use SSE4.1 instructions, but these are available on your machine and could speed up CPU computations.
2017-05-22 21:08:30.215110: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use SSE4.2 instructions, but these are available on your machine and could speed up CPU computations.
2017-05-22 21:08:30.215115: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use AVX instructions, but these are available on your machine and could speed up CPU computations.
2017-05-22 21:08:30.215119: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use AVX2 instructions, but these are available on your machine and could speed up CPU computations.
2017-05-22 21:08:30.215123: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use FMA instructions, but these are available on your machine and could speed up CPU computations.
############################### distance = 58.55 meters
('############################### FPS = ', 1471.914922886232)
ended benchmark
1000
Writing 50286 timings for thread 0
Writing 68342 timings for thread 1
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
```

erwincoumans, 2017-05-23:
great, it works! You can enable the 'sleep' command to see what is happening. I haven't spend time to make it a nice looking / running demo.

ahundt, 2017-05-23:
cool I'd say this issue can be closed now. Thanks!

closed 2017-05-23 by ahundt