bulletphysics/bullet3 #1148: Segmentation fault on pybullet loadURDF with floating joint

tpbarron, 2017-05-25:
I'm trying to load a sphere with a floating joint so that velocity control can be used. Pybullet gives an error on loadURDF:

> Thread 14 "python" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc3173700 (LWP 2304)]
__memcpy_avx_unaligned () at ../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S:114
114	../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S: No such file or directory.

If I switch to a fixed joint everything loads fine. Here is basic code to reproduce:

```
import pybullet as p

PLANE_PATH = 'res/plane/plane.urdf'
SPHERE_PATH = 'res/sphere2/sphere2_rolling_friction_jointed.urdf'

physicsClient = p.connect(p.GUI)
p.setGravity(0, 0, -9.8)
p.setRealTimeSimulation(0)
planeId = p.loadURDF(PLANE_PATH, [-0.5, -0.5, 0])
print ("loaded plane")
obj = p.loadURDF(SPHERE_PATH, [0.0, 0.0, 0.5])
print ("loaded sphere")
print (p.getNumJoints(obj))

while True:
    p.stepSimulation()
```

I am using newly updated pybullet 1.0.8. Is there a simple fix for this? Thanks.

[sphere2_rolling_friction_jointed.urdf.txt](https://github.com/bulletphysics/bullet3/files/1029715/sphere2_rolling_friction_jointed.urdf.txt)

erwincoumans, 2017-05-25:
We don't support a floating, spherical or planar joints at the moment. We only have a motorized prismatic and revolute joint. We should fix a crash for unsupported joints though. You could try to apply forces/torques instead?

erwincoumans, 2017-05-25:
crash fixed here:: https://github.com/bulletphysics/bullet3/pull/1149
it will create a fixed joint instead of crashing. Bullet/pybullet will automatically create a floating base, unless the object has zero mass (fixed) or attached to the world using a prismatic or revolute joint.

closed 2017-05-26 by tpbarron

erwincoumans, 2017-05-26:
Can you explain what you try to achieve? If you want to control the linear and/or angular velocity of the sphere, with respect to a fixed world frame, we could introduce some kind of special motor constraint. (that you can create using pybullet.addConstraint(...).

tpbarron, 2017-05-26:
I'm trying to create a simple maze-like RL environment and was going to start with a spherical agent for simplicity. I'd like to be able to control the linear velocity of the sphere with respect to the world frame. 
So it should work work to use a motor constraint between the sphere and a fixed point on a plane, right? Thanks for your help.

erwincoumans, 2017-05-26:
It probably makes most sense to control only the angular velocity, and leave the friction transfer that to linear velocity etc. Since we don't have this constraint, you may want to use applyTorque instead. Then, you need to use 'setRealTimeSimulation(0), and apply this torque before each call to stepSimulation.