bulletphysics/bullet3 #1154: How do you generate playback data for kuka_grasp_block_playback.py

ahundt, 2017-05-28:
I went to try `kuka_grasp_block_playback.py`, but the playback file appears the only file where the kuka with gripper sdf is loaded. How do I try out that example/generate data for it?

erwincoumans, 2017-05-28:
We haven't created an example to generate data, but that should be pretty easy.
Basically, you need to create/load the same initial objects from URDF/SDF, then start the generic logging of all object: p.startStateLogging(p.STATE_LOGGING_GENERIC_ROBOT, "logFile.bin").

ahundt, 2017-05-29:
Okay, I was now able to get the data log to run, I just had to symlink the `bullet3/data` folder into `bullet3/examples/pybullet/examples` to resolve this error:

```
± python kuka_grasp_block_playback.py
pybullet build time: May 28 2017 13:19:59
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce GT 750M OpenGL Engine
Version: 4.1 NVIDIA-10.10.14 310.42.25f02
GLSL: 4.10
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0
MotionThreadFunc thread started
Traceback (most recent call last):
  File "kuka_grasp_block_playback.py", line 58, in <module>
    log = readLogFile("data/block_grasp_log.bin")
  File "kuka_grasp_block_playback.py", line 14, in readLogFile
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: 'data/block_grasp_log.bin'
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
destroy semaphore
semaphore destroyed
Thread with taskId 0 exiting
Thread TERMINATED
destroy main semaphore
```

If that should run by default could update the path in the python file, beyond that this issue can be closed.

erwincoumans, 2017-05-29:
I fixed the data folder, and also remove many unused imports, see this [commit ](https://github.com/bulletphysics/bullet3/pull/1155/commits/5436b8f048887b8f2c86059442a1041d076ded4b)in pull request https://github.com/bulletphysics/bullet3/pull/1155.

ahundt, 2017-05-29:
pulled, it works!

closed 2017-05-29 by ahundt