bulletphysics/bullet3 #1156: sdf + urdf examples may not comply with their format specs

ahundt, 2017-05-28:
I tried to load the [data/kuka_iiwa](https://github.com/bulletphysics/bullet3/tree/master/data/kuka_iiwa) models as detailed in this [v-rep forum post](http://www.forum.coppeliarobotics.com/viewtopic.php?f=5&t=6712) and in the latest gazebo 8.1:
```
○ /usr/local/Cellar/gazebo8/8.1.0/bin/gazebo /Users/athundt/source/bullet3/data/kuka_iiwa/kuka_with_gripper.sdf
Warning [parser.cc:781] XML Element[contact], child of element[link] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:781] XML Element[contact], child of element[link] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
```

I also tried kuka_world.sdf in gazebo 8.1:
```
○ /usr/local/Cellar/gazebo8/8.1.0/bin/gazebo /Users/athundt/source/bullet3/data/kuka_iiwa/kuka_world.sdf
2017-05-28 17:52:16.111 gzclient[35216:565505] Layout still needs update after calling -[QNSPanelContentsWrapper layout].  QNSPanelContentsWrapper or one of its superclasses may have overridden -layout without calling super. Or, something may have dirtied layout in the middle of updating it.  Both are programming errors in Cocoa Autolayout.  The former is pretty likely to arise if some pre-Cocoa Autolayout class had a method called layout, but it should be fixed.
```

In all cases that the file loaded, the meshes didn't seem to display and often the files didn't load either. Since it fails in two third party tools, that leads me to guess that some of these models may not comply with the format properly.

This is tangentially related to #1154, since I'm trying to try out the same models with different tools.

erwincoumans, 2017-05-28:
In V-Rep, that is just a warning for a custom element we added (contact), that should just be ignored. What are the actual errors in the URDF file? In gazebo, it seems some unrelated error about its layout.
If you find the actual issues, we can look into fixing the URDF.

ahundt, 2017-05-28:
I know this is borderline out of scope, so I can close this if that's the case. 

### gazebo

Those warnings were for gazebo and while there were only warnings, no objects or models were visible in the scene. I guess this could be due to the obj format as you mentioned?

this gazebo command: 

```
○ /usr/local/Cellar/gazebo8/8.1.0/bin/gazebo /Users/athundt/source/bullet3/data/kuka_iiwa/kuka_world.sdf
2017-05-28 19:38:30.412 gzclient[43382:738631] Layout still needs update after calling -[QNSPanelContentsWrapper layout].  QNSPanelContentsWrapper or one of its superclasses may have overridden -layout without calling super. Or, something may have dirtied layout in the middle of updating it.  Both are programming errors in Cocoa Autolayout.  The former is pretty likely to arise if some pre-Cocoa Autolayout class had a method called layout, but it should be fixed.
```
Produced this scene with no objects:

[image]

### v-rep

Sorry my post was a bit confusing regarding V-REP, the V-REP errors are at the [v-rep forum post link](http://www.forum.coppeliarobotics.com/viewtopic.php?f=5&t=6712), which I've now reproduced here:

I tried to load the urdf and sdf files provided for bullet3 example data:
https://github.com/bulletphysics/bullet3/tree/master/data/kuka_iiwa

When I try to load kuka_with_gripper.sdf I get the following error:
```
<sdf>: <world>: sub element gravity not found
```

In model.urdf vrep seems to choke on the relative paths to the meshes:
```
URDF import operation started.
There are 7 joints.
ERROR: there is no joint with name ''
There are 8 links.
Creating link 'lbr_iiwa_link_0'...
ERROR: neither mesh file 'meshes/link_0.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_0.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_0.stl' nor '' do exist.
ERROR: failed to create the mesh 'meshes/link_0.stl' or '' with extension type 4
ERROR: there is no joint with name ''
Creating link 'lbr_iiwa_link_1'...
ERROR: neither mesh file 'meshes/link_1.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_1.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_1.stl' nor '' do exist.
ERROR: failed to create the mesh 'meshes/link_1.stl' or '' with extension type 4
Creating link 'lbr_iiwa_link_2'...
ERROR: neither mesh file 'meshes/link_2.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_2.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_2.stl' nor '' do exist.
ERROR: failed to create the mesh 'meshes/link_2.stl' or '' with extension type 4
Creating link 'lbr_iiwa_link_3'...
ERROR: neither mesh file 'meshes/link_3.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_3.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_3.stl' nor '' do exist.
ERROR: failed to create the mesh 'meshes/link_3.stl' or '' with extension type 4
Creating link 'lbr_iiwa_link_4'...
ERROR: neither mesh file 'meshes/link_4.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_4.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_4.stl' nor '' do exist.
ERROR: failed to create the mesh 'meshes/link_4.stl' or '' with extension type 4
Creating link 'lbr_iiwa_link_5'...
ERROR: neither mesh file 'meshes/link_5.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_5.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_5.stl' nor '' do exist.
ERROR: failed to create the mesh 'meshes/link_5.stl' or '' with extension type 4
Creating link 'lbr_iiwa_link_6'...
ERROR: neither mesh file 'meshes/link_6.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_6.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_6.stl' nor '' do exist.
ERROR: failed to create the mesh 'meshes/link_6.stl' or '' with extension type 4
Creating link 'lbr_iiwa_link_7'...
ERROR: neither mesh file 'meshes/link_7.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_7.stl' nor '' do exist.
ERROR: neither mesh file 'meshes/link_7.stl' nor '' do exist.
ERROR: failed to create the mesh 'meshes/link_7.stl' or '' with extension type 4
There are 0 sensors.
URDF import operation finished.

```

According to the SDF definition, gravity is optional:
http://sdformat.org/spec?ver=1.6&elem=link#link_gravity

Also, I tried putting the stl mesh files in the exact same folder as the urdf (rather than in the meshes folder) on the assumption that there was a  and the files were still not found.

In case this is the fault of the models themselves, I created a corresponding issue in the bullet3 github repo:

https://github.com/bulletphysics/bullet3/issues/1156

erwincoumans, 2017-05-29:
Those KUKA models don't use .OBJ, so that's not the problem. You may want to experiment with a really small and simple working URDF/SDF in both Gazebo and V-REP first. Then modify the working URDF/SDF step-by-step to the Bullet KUKA URDF/SDF until you track down where things fail in V-REP and Gazebo.

erwincoumans, 2017-05-30:
Do you have any very small URDF file (with assets) of a textured cube (with visual cube and collision cube) that loads in V-REP and Gazebo?

ahundt, 2017-05-30:
I think our ur5 model mentioned in https://github.com/bulletphysics/bullet3/issues/1109 might meet the requirements, but with `package://` though I think the files could just be copied to a local spot and updated. @cpaxton do you think that might work or perhaps know of another example?

cpaxton, 2017-05-30:
That urdf works in gazebo perfectly well, although it now looks pretty ugly because i converted the DAEs to OBJs imperfectly. There is some issue with use of `package://` and wavefront mtl textures that i have not yet resolved.

cpaxton, 2017-05-30:
Note that we do use xacro as well but only to compile -- you can copy the final URDF and start modifying that. I have noticed problems importing urdfs into Bullet as well and would like to see this resolved.

erwincoumans, 2017-05-30:
Well, I'm hoping to start with a simple textured cube, not a complex UR5 robot, that loads fine in Gazebo and V-REP, preferably using a textured triangle meshes (no complex COLLADA non-triangulated polyhedra).

ahundt, 2017-05-30:
I bet the easiest way to create a model would be to make it in gazebo, the brick box might fit the bill, I'm not sure if the the collada file includes the texture but it is in the default gazebo 8.1 install:

[image]

[simple_box.dae](http://models.gazebosim.org/brick_box_3x1x3/meshes/simple_box.dae)

SDF file exported by gazebo 8.1:

```
<sdf version='1.6'>
  <world name='default'>
    <light name='sun' type='directional'>
      <cast_shadows>1</cast_shadows>
      <pose frame=''>0 0 10 0 -0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.1 -0.9</direction>
    </light>
    <model name='ground_plane'>
      <static>1</static>
      <link name='link'>
        <collision name='collision'>
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>100 100</size>
            </plane>
          </geometry>
          <surface>
            <contact>
              <collide_bitmask>65535</collide_bitmask>
              <ode/>
            </contact>
            <friction>
              <ode>
                <mu>100</mu>
                <mu2>50</mu2>
              </ode>
              <torsional>
                <ode/>
              </torsional>
            </friction>
            <bounce/>
          </surface>
          <max_contacts>10</max_contacts>
        </collision>
        <visual name='visual'>
          <cast_shadows>0</cast_shadows>
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>100 100</size>
            </plane>
          </geometry>
          <material>
            <script>
              <uri>file://media/materials/scripts/gazebo.material</uri>
              <name>Gazebo/Grey</name>
            </script>
          </material>
        </visual>
        <self_collide>0</self_collide>
        <enable_wind>0</enable_wind>
        <kinematic>0</kinematic>
[... 93 lines cut ...]
          <wrench>0 0 0 0 -0 0</wrench>
        </link>
      </model>
      <light name='sun'>
        <pose frame=''>0 0 10 0 -0 0</pose>
      </light>
    </state>
    <gui fullscreen='0'>
      <camera name='user_camera'>
        <pose frame=''>13.4614 -11.9129 4.3509 0 0.275643 2.35619</pose>
        <view_controller>orbit</view_controller>
        <projection_type>perspective</projection_type>
      </camera>
    </gui>
  </world>
</sdf>

```

It also seems [matlab has a simple urdf example](https://www.mathworks.com/help/physmod/sm/ug/urdf-model-import.html), though I don't think it has a texture, and they claim [solidworks now supports urdf export](https://www.mathworks.com/help/physmod/sm/ug/urdf-model-import.html#bvmylh6-1):

>Obtaining URDF Models to Import
You can, but generally do not have to, manually create your own URDF files. For more complex models, it can be preferable to obtain URDF files from other sources. Robotics manufacturers and consultants often provide URDF models for their robotic systems. CAD applications such as SolidWorks® and PTC® Creo™ support URDF exporters that convert your CAD assemblies into URDF models. Consider these options when working with complex robotics models that may not be simple to create manually.

As an aside I wanted to mention the [SDFormat library](sdformat.org) ([bitbucket hg repo](https://bitbucket.org/osrf/sdformat) and [4.0 api](http://osrf-distributions.s3.amazonaws.com/sdformat/api/4.0.0/hierarchy.html)). It looks like it uses tinyxml as the only dependency I could find, which I think bullet uses too. It has a built in urdf parser and is licensed under Apache v2.

erwincoumans, 2017-05-30:
The matlab file loads just fine. It seems the main incompatibility is due to graphics assets: Bullet only support triangle meshes in COLLADA dae files and mainly uses Wavefront .obj. V-Rep and Gazebo don't support Wavefront .obj? So if you really want a common sub-set, you will have to stick to COLLADA triangulated meshes. 

Bullet can load the attached URDF + COLLADA .dae. Can V-Rep and Gazebo load it too (aside from lacking texture)?
[simple_box.zip](https://github.com/bulletphysics/bullet3/files/1039883/simple_box.zip)

Once that works, we can look into adding a texture, preferably in the same directory as the urdf and dae file.

ahundt, 2017-05-31:
In V-REP the model loaded but the texture did not.

It seems all urdf files must be converted to SDF for gazebo and the documentation appears out of date so I created the following issue on their repo to figure it out.
https://bitbucket.org/osrf/gazebo/issues/2289/how-to-convert-urdf-to-sdf-and-load-it

erwincoumans, 2017-05-31:
Having a perfect URDF+SDF+COLLADA+STL+image format inter-compatibility takes a huge effort, and there are likely higher priority issues for all of the simulator-creators (Gazebo, V-Rep, pybullet, Matlab etc).

Still, it would be nice to have a couple of simple reference URDF and SDF files, including DAE triangle meshes and texture images, and push the creators to fix compatibility :)

ahundt, 2017-06-05:
Thanks to @fferri, [he confirmed on the vrep forum](http://www.forum.coppeliarobotics.com/viewtopic.php?f=5&t=6712#p26754) that some SDF specs aren't being met in [kuka_with_gripper.sdf](https://github.com/bulletphysics/bullet3/blob/master/data/kuka_iiwa/kuka_with_gripper.sdf).

In particular at least one error is that the [world gravity tag isn't present as required](http://sdformat.org/spec?ver=1.6&elem=world#world_gravity).

ahundt, 2017-06-05:
I'm sure complete compatibility is unlikely, but perhaps it is reasonable for a few basic robot model & simulation examples so people might have a chance of moving a subset of their data between tools. :-)

erwincoumans, 2017-06-05:
Please help and modify our sdf and or urdf, until it works with v-rep and
gazebo. We can consider some improvements to Bullet's Sdf/urdf/ dae parsers.

ahundt, 2017-06-06:
Sure, I'll try that. Hopefully it won't be a rabbit hole of one small fix after another. My first steps and the [bitbucket issue](https://bitbucket.org/osrf/gazebo/issues/2289/how-to-convert-urdf-to-sdf-and-load-it) in the previous posts were in the hopes of an automated conversion to create the reference file.

ahundt, 2017-06-06:
Okay I think I made some progress with conversion... loading things up in gazebo isn't exactly a walk in the park it seems they need things to be in a very specific layout which I detail below.

```
└── simple_box
    ├── model.config
    ├── simple_box.dae
    ├── simple_box.sdf
    └── simple_box.urdf

```

The following gazebo command will convert files, provide warnings about what is wrong, and print a converted file out on the command line: 

```bash
gz-8.1.1 sdf -p ~/src/bullet3/data/kuka_iiwa/kuka_with_gripper.sdf
```

The printout which can be redirected to overwrite the original sdf file, for example:

```bash
gz-8.1.1 sdf -p ~/src/bullet3/data/kuka_iiwa/kuka_with_gripper.sdf >  ~/src/bullet3/data/kuka_iiwa/kuka_with_gripper.sdf 
```

### Models

Here are some very simple models I've compressed for convenience, plus an updated `simple_box.zip`, these all load in both gazebo and V-REP except V-REP doesn't seem to support textures:
- [simple_box.zip](https://github.com/bulletphysics/bullet3/files/1053603/simple_box.zip) your simple_box updated to load in gazebo and V-REP
- [ball_bearing.zip](https://github.com/bulletphysics/bullet3/files/1053568/ball_bearing.zip) should be dark gray, has a material definition (a V-REP option), and is otherwise very simple
- [bookshelf.zip](https://github.com/bulletphysics/bullet3/files/1053569/bookshelf.zip) includes a texture and is composed of several simple boxes
- [cinder_block.zip](https://github.com/bulletphysics/bullet3/files/1053606/cinder_block.zip) includes a collada mesh

The materials spec they have also seems to be important if there is to be a shelf made of wood:
```

        <material>
          <script>
            <uri>file://media/materials/scripts/gazebo.material</uri>
            <name>Gazebo/Wood</name>
          </script>
        </material>
```

And gazebo won't load model/mesh files next to the sdf file unless it is specified with `model://model_folder_name/model_mesh_file.dae`, as you can see in this working version of `simple_box.dae`:

```xml
<sdf version='1.6'>
  <model name='cube'>
    <link name='baseLink'>
      <pose frame=''>0 0 0 0 -0 0</pose>
      <inertial>
        <pose frame=''>0 0 0 0 -0 0</pose>
        <mass>1</mass>
        <inertia>
          <ixx>1</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>1</iyy>
          <iyz>0</iyz>
          <izz>1</izz>
        </inertia>
      </inertial>
      <collision name='baseLink_collision'>
        <pose frame=''>0 0 0 0 -0 0</pose>
        <geometry>
          <box>
            <size>1 1 1</size>
          </box>
        </geometry>
      </collision>
      <visual name='baseLink_visual'>
        <pose frame=''>0 0 0 0 -0 0</pose>
        <geometry>
          <mesh>
            <scale>1 1 1</scale>
            <uri>model://simple_box/simple_box.dae</uri>
          </mesh>
        </geometry>
      </visual>
    </link>
  </model>
</sdf>
```

It seems part of the solution is that the models don't need to define the `world`, they can simply be a `model` like the files above. An additional requirement for gazebo is the `model.config` file, which specifies the name of the object and the path to the sdf file. Gazebo also requires model file (like collada).

### Simulations

The `world` definition is what has all those extra requirements such as specifying gravity, light sources, simulation refresh rate, etc. You may want to support simulation definitions, but it should likely not be used for simple model definitions.

Here is everything needed for a basic simulation world definition but note that there seems to be a [small bug in the saved vs loaded dimensions as specified in gazebo 8.1.1](https://bitbucket.org/osrf/gazebo/issues/2292/world-does-not-save-and-load-consistently).

- [brick_box_world.zip](https://github.com/bulletphysics/bullet3/files/1053693/brick_box_world.zip) A world containing default settings and a brick box
- [brick_box_3x1x3.zip](https://github.com/bulletphysics/bullet3/files/1053694/brick_box_3x1x3.zip) the brick box model in the above world

gavanderhoorn, 2017-06-30:
@erwincoumans wrote:
> I think URDF and SDF would have been better with Wavefront .obj support (and STL) instead of COLLADA.

perhaps little known, but ROS (or at least RViz) can actually load whatever `assimp` can load. Even `.fbx`.

ahundt, 2017-07-04:
> perhaps little known, but ROS (or at least RViz) can actually load whatever `assimp` can load. Even `.fbx`.

Interesting suggestion, something like that could probably help provide broader support with fewer implementation steps, though I think that would likely need to be in addition to URDF and SDF. Probably would need a volunteer for implementation though, haha.

erwincoumans, 2017-07-04:
>>in addition to URDF and SDF. 

It is not "in addition": URDF and SDF points to triangle meshes in COLLADA and STL format. If Gazebo uses assimp, it would allow URDF/SDF files to point to Wavefront .OBJ files (like Bullet allows).

gavanderhoorn, 2017-07-04:
@ahundt: it's as @erwincoumans writes: assimp can load quite some mesh formats. But they'll be used as geometry only.

I just wanted to point out that loading wavefront meshes probably already works.

erwincoumans, 2018-12-23:
Closing issue. If a URDF doesn't load well, check following:
+ use obj/stl instead of dae meshes
+ use relative folders for meshes, instead of 'package://' or 'model://' etc.

closed 2018-12-23 by erwincoumans