bulletphysics/bullet3 #1157: OpenGL rendering in headless mode is broken

jaredweiss, 2017-05-29:
Using the rendering option `ER_BULLET_HARDWARE_OPENGL` in `getCameraImage` while using a  `DIRECT` pybullet server (i.e. no GUI) will render/return blank canvases. If headless rendering shouldn't be possible with the opengl renderer an error should be thrown instead of just returning an empty image.

erwincoumans, 2017-05-29:
well, DIRECT mode implies no GUI and no OpenGL, so you can obviously not use OpenGL to render images. Use ER_TINY_RENDERER instead. pybullet doesn't have verbose error messages indeed.

closed 2017-05-29 by erwincoumans

jaredweiss, 2017-05-29:
got it. thanks

joemathai, 2017-06-27:
I am facing a similar issue, ER_TINY_RENDERER is returning a blank canvas in DIRECT mode. But using the same setup with ER_BULLET_HARDWARE_OPENGL in GUI mode returns the correct image. Am I missing something?

erwincoumans, 2017-06-27:
It works just fine with out scenes, it is likely due to your URDF files.
Did you try running the Bullet/examples/pybullet/examples/testrender.py script? It uses TinyRenderer by default.

carismoses, 2019-06-24:
I think that there is a bug in rendering option ER_TINY_RENDERER. To properly render in DIRECT mode, you have to use the viewMatrix and projectionMatrix options when you call getCameraImage. In the Bullet/examples/pybullet/examples/testrender.py script, if you remove either of these options from the getCameraImage call (line 49), the images don't render properly.