bulletphysics/bullet3 #1159: Flying human in tensorflow/humanoid_running.py

cprogrammer1994, 2017-05-30:
Running the example:

[image]

Setting gravity:

```py
p.setGravity(0,0,-9.8 * 1.5)
```

[image]

This happens with 50% chance

benelot, 2017-05-30:
> Running the example:
>
> [image: tfexamplebullet]
> <https://cloud.githubusercontent.com/assets/11232402/26581676/0d44778e-453e-11e7-9c75-43ff344073b2.gif>
>
> Setting gravity:
>
> p.setGravity(0,0,-9.8 * 1.5)
>
> [image: pybullettensorflow_gravity_x1_5]
> <https://cloud.githubusercontent.com/assets/11232402/26581708/3376e388-453e-11e7-860a-d7eab363fef8.gif>
>
> This happens with 50% chance

The trained neural network seems to have learned a policy which only makes
sense when in contact with ground and with -9.81 as the gravity. Probably
the additional gravity you add pulls the humanoid to ground, which it tries
to compensate with more force to lift up again, so it overshoots since it
is not trained for this and jumps, thus starts flying into the air. And
when above ground, since it was not trained for that either, it performs
pretty random movements, which are too high to be simulated properly. One
could train the humanoid with varying gravity to improve the situation.
I am currently working on making training available for such models. Then
you can improve the network by yourself.

cprogrammer1994, 2017-05-30:
gravity 0.5->1.2 seems to work fine

cprogrammer1994, 2017-05-30:
Is the data after line 200 trained somewhere else in the examples?

benelot, 2017-05-30:
For now, there are no other trained networks. But soon there are more
examples to come like it.

cprogrammer1994, 2017-05-30:
Great, thanks!

erwincoumans, 2018-12-23:
Should be fixed by now. Thanks!

closed 2018-12-23 by erwincoumans