bulletphysics/bullet3 #1163: Pybullet Backend API for Physics & Render Engine Independence

ahundt, 2017-05-31:
I like the philosophy and approach of [François Chollet with Keras in TensorFlow](https://www.youtube.com/watch?v=UeheTiBJ0Io) described in the first minute of the linked video, which focuses on reducing complexity and improving ease of use.

While replying to #1138 I realized a backend design might help meet the independence goals of pybullet described in the [pybullet quick start introduction](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3). I think a complete design would also need at least some of the geometry, linear algebra, realtime fwd+inv kinematics & dynamics described in #1138.

**Pybullet Backend API for independence from the underlying physics and render engine**

Keras provides a good reference design for [multiple backend compatibility](https://keras.io/backend/#keras-backends) (also see the [backend source code link](https://github.com/fchollet/keras/tree/master/keras/backend)) which supports TensorFlow, Theano, and will soon support CNTK. Here is an example from the [Keras backend docs](https://keras.io/backend/):

```python
from keras import backend as K

# This is creating tensors, but it could just as well be immediate execution
b = K.random_uniform_variable(shape=(3, 4)). # Uniform distribution
c = K.random_normal_variable(shape=(3, 4)). # Gaussian distribution
d = K.random_normal_variable(shape=(3, 4)).

# all-zeros variable:
var = K.zeros(shape=(3, 4, 5))
# all-ones:
var = K.ones(shape=(3, 4, 5))

a = b + c * K.abs(d)
c = K.dot(a, K.transpose(b))
a = K.sum(b, axis=1)
```

The backend concept could be adapted in three ways (1a, 1b, 2):

1. Create an independent pybullet backend inspired by Keras backends with either:
    a. Immediate execution (standard python / numpy / pybullet style execution)
    b. Tensor graph based execution (tensorflow/theano/keras style execution)
2. Adopt and extend the actual current keras backend API for pybullet.

In addition to a backend, do consider tensor graphs for the performance benefits, debugging tools, and easy integration with TF & friends. :-)

erwincoumans, 2017-05-31:
We can already have different backends for pybullet, replacing Bullet 2.x and TinyRenderer with a different physics engine (or Bullet 3.x) and different renderer. (replacing the PhysicsServerCommandProcessor). This would require some C++ code.

I think pybullet is separate from the machine learning frameworks (Tensorflow), there is no intention to create a dependency between pybullet and Tensorflow.

ahundt, 2017-05-31:
Please allow me to retry my explanation in a much broader way I hope will be more clear. I was thinking about the philosophical design principles and the Keras design to support backends in a software engineering sense. In other words, I'm proposing we pretend Keras isn't an ML library for a moment, and focus on general API design concepts. :-) 

The guiding principles of Keras, as [François explains in the first minute of the youtube video](https://www.youtube.com/watch?v=UeheTiBJ0Io), are also [on the top of the main keras.io page](https://keras.io/). They could, in theory, be adapted to PyBullet as follows:

- **User friendliness**. Pybullet is an API designed for human beings, not machines. It puts user experience front and center. PyBullet follows best practices for reducing cognitive load: it offers consistent & simple APIs, it minimizes the number of user actions required for common use cases, and it provides clear and actionable feedback upon user error.
- **Modularity**. A physics simulation model can be understood as either a sequence or a graph of standalone, fully-configurable modules that can be plugged together with as few restrictions as possible. In particular the renderer, linear algebra engine, physics engine, initialization schemes, Inverse Kinematics cost functions and optimizers are all standalone modules that you can combine to create new simulation models.
- **Easy extensibility**. New modules are simple to add (as new classes and functions), and existing modules provide ample examples. To be able to easily create new modules allows for total expressiveness, making PyBullet suitable for advanced physics simulation and research.
- **Work with Python**. No separate physics models, configuration files in a declarative format. Models are described in Python code, which is compact, easier to debug, and allows for ease of extensibility.

> We can already have different backends for pybullet, replacing Bullet 2.x and TinyRenderer with a different physics engine (or Bullet 3.x) and different renderer. (replacing the PhysicsServerCommandProcessor). This would require some C++ code.

Cool perhaps those backends could be selected in the same manner as they are in Keras? 

Here is an example of [keras.json](https://keras.io/backend/#kerasjson-details) adapted to a hypothetical `bullet3.json`:

```json
{
    "backend": 
     {
        "physics": "bullet3",
        "linear_algebra": "eigen3",
        "rendering": "tiny_renderer"
     }
}
```

Plus an adapted example determining which backend is in use:

```python
# options could be specified in bullet3.json
# or before bullet3 is imported

# linalg options could be 'bullet3', 'eigen3'
>>> pybullet.linalg.backend.backend()
'bullet3'

# render options could be 'tinyrenderer', 'blender_cycles'
>>> pybullet.rendering.backend.backend()
'tinyrenderer'

# additional physics options could be tbd
>>> pybullet.physics.backend.backend()
'bullet3'
```

> I think pybullet is separate from the machine learning frameworks (Tensorflow), there is no intention to create a dependency between pybullet and Tensorflow.

I definitely agree that this is the right way to go, what I'm trying to say is that with backends the implementations for physics, linear algebra, and rendering could be selected. 

An additional side note is that the default linear algebra backend of PyBullet could create a simple graph of math operations which, when run, does those calculations with the selected backend, should it be bullet3, eigen, or something else entirely in C++ for performance reasons. The Keras backend API (with ML specific calls removed) could provide inspiration for such a mechanism. ML backends could very well be available, but this isn't necessary at all and should definitely not be a dependency.

erwincoumans, 2017-05-31:
From your posting you seem to think we consider large changes in approach, but I think pybullet is useful as-is and getting incrementally in a nicer shape based on feedback from users who are really using it.

I wouldn't expect large changes in the current API, and there will be no serious linear algebra functionality, aside from the few functions we have. It seems more constructure to discuss new features or bugs/issues in the current pybullet, rather than implying big changes in the current pybullet. If you are running into some real issues, please reproduce your issue in a pybullet reproduction case.

There is no need for json complexity to switch physics and/or renderer backend:

1) you connect to the physics backend through pybullet.connect(method=...).
2) the renderer can be selected through the getCameraImage(..., renderer=...)

Once we have a new physics engine backend, there will simply be one or more new method enums to choose from.

For the renderer, we have ER_TINY_RENDERER (CPU/software) and ER_BULLET_HARDWARE_OPENGL.
We have already some external renderer bridges, outside of pybullet, that would work well too, for Cycles etc.

erwincoumans, 2017-06-05:
Since the pybullet design already supports multiple renderers (both on physics server side and on client side) and also multiple physics backends (through deriving from PhysicsCommandProcessorInterface and selecting through the connect method) this can be closed.

closed 2017-06-05 by erwincoumans