bulletphysics/bullet3 #1175: Documentation for softbody objects in PyBullet

cpaxton, 2017-06-08:
No documentation or existing tools on creating softbody objects from pybullet. It looks like they might need to be created via C++. More documentation would be helpful.

erwincoumans, 2017-06-08:
Softbody hasn't been a priority, there was just a small test (soft body bunny) for grasping research. In the C-API, there are some b3LoadBunny* commands we could trivially hook up to pybullet. For this to be useful, you likely want to allow loading soft bodies from some standard file format (tetgen tegrahedral meshes or so).

cpaxton, 2017-06-08:
I was interested in creating soft primitives (specifically the spheres or toruses in the example browser), but at different scales. Sounds like this would be relatively easy to expose in pybullet?

airballking, 2017-09-14:
I would also be interested in using soft bodies from pybullet. Has there been any progress on this issue?

Specifically, I am looking for ways to simulate something like pancakes that can fold together or partially hang down a corner. The initial shape would come from some mesh file. It seems like the C++ API would allow doing this. Is this correct?

I would also be willing to invest some time in preparing a pull request. @erwincoumans would you be interested in such a feature request, and would you give some guidance in wrapping the existing C++ features?

erwincoumans, 2017-09-14:
There is some super experimental softbody support in the Bullet C-API, which is underlying pybullet. See for example [b3LoadBunnyCommandInit](https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/PhysicsClientC_API.h#L448) in PhysicsClientC_API.h. THere is basic btSoftBody versus btMultiBody.

It would be easy to hook it up to pybullet. Better would be to generalize beyond bunnies :-) Mind you, the pybullet.* code and wrapper classes are not pretty (read messy) to work with. Still, you could give it a try, experiment a bit further with it.

ugocupcic, 2017-10-06:
@erwincoumans I'm pretty interested in spawning softbodies from pybullet directly for simulating grasping as well.

In one of your comments above, you state: "we ended up just using rigid bodies with some compliance parameters (stiffness/damping)". Was it sufficient to get "realistic" grasps?

Cheers,

erwincoumans, 2017-10-09:
We got reasonably well transfer of grasps from a model trained in sim, mixed with some episodes trained on real hardware. There is not much deformation happening in our grippers, and non-slip and lateral+torsional friction with some compliance worked well. We will get to softbody and fem, but need to add the same quality contact and friction under larger deformation.

githubkusi, 2019-07-09:
I would be interested in modeling a rope in PyBullet (as seen in examples/SoftDemo/SoftDemo.cpp)

DarioUrbina, 2020-07-13:
I am also interested in the Rope example, does it exist for pyBullet?

I am currently simulating many bodies connected in series of revolute joints, this seems to generate a rope-like behavior, But perturbing such a "rope" with fast force variations generated a chaotic behavior of the simulation. I guess this is because the time steps don't allow bullet to solve all the constraints involved. 

I have also tried using the createContraint function to create connections of different urdf files through python code. These constraints, even though they are declared as "fixed", appear to change. 

Any suggestions?

closed 2021-04-26 by erwincoumans