bulletphysics/bullet3 #1188: Why human robot flying in tensorflow/humanoid_running.py ?

foolyc, 2017-06-15:
the original py file ensorflow/humanoid_running.py:
[image]

even when i set all the joint force to 0
`		forces = [0.] * len(motors)
		for m in range(len(motors)):
			forces[m] = 0#motor_power[m]*actions[m]*0.082
		p.setJointMotorControlArray(human, motors,controlMode=p.TORQUE_CONTROL, forces=forces)`
the human robot still flies.
so whats the problem?

erwincoumans, 2017-08-09:
What operating system (version etc) and Python version and pybullet version are you using? Does the unmodified humanoid_running.py have the issue?

erwincoumans, 2018-12-23:
Should work better in latest version. Closing issue.

closed 2018-12-23 by erwincoumans