bulletphysics/bullet3 #1192: Need documentation for updating to Bullet 2.86

LiamM32, 2017-06-16:
I'm trying to update the codebase for [Lips of Suna](https://gitlab.com/xenodora/lipsofsuna) to support Bullet 2.86.

Since [this commit](https://github.com/bulletphysics/bullet3/commit/1c3686ca51866308d08c97a747eb34a7c450b0c1), _m_simplexSolver_ & _getSimplexSolver()_ have been removed from _btDefaultCollisonConfiguration_.  This is what is preventing Lips of Suna from building with Bullet 2.86.  I am trying to update it, but the documentation is lacking so I can't figure-out how to update it properly.  I can't find information on achieving the same functionality in Bullet 2.86.

It would be nice if API changes were documented similarly to how they are for [Blender scripts](https://wiki.blender.org/index.php/Extensions:2.6/Py/API_Changes).

lunkhound, 2017-06-17:
It turned out to be problematic to have a global simplexSolver object when doing collision detection on multiple threads. So each thread creates a btVoronaiSimplexSolver on the stack whenever it needs one.
So there is no longer any need to pass a simplex solver into the ConvexConvexAlgorithm constructor anymore.
Basically you should not be passing around a pointer to a global simplexSolver object anymore.

LiamM32, 2017-06-17:
So I just remove it? Nothing else?  I was able to get the program to compile by doing that, but it segfaults immediately after starting-up.
As you can tell, I am not experienced in C++.

lunkhound, 2017-06-17:
Yes, removing it should do the trick.

I can only guess what could be causing your segfault. One thing you may want to check is if your codebase  has any classes deriving from Bullet classes that are overriding virtual functions. If any of those virtual functions have changed their signature since your previous version of Bullet then you could have something that compiles fine but is broken.

There is an easy way to turn those into compiler errors -- just mark all the functions in your code that are supposed to be overriding a Bullet virtual function with the "override" keyword. Then the compiler should give you an error for every such function that doesn't match with something in the parent class.
It's a C++11 keyword though, so you may need to enable it in your compiler.

LiamM32, 2017-06-20:
Thank you for helping me with this.  I tried putting "override" on many functions, including the one that gdb is pointing fingers at.  The project uses _waf_ to compile, so I don't know how to enable C++11.

I should note that this is not a project that I made.  It's abandoned by it's creator, but I'm trying to pick it up.

closed 2021-04-26 by erwincoumans