bulletphysics/bullet3 #1197: Wrong depth values using Tiny Renderer

mbreyer, 2017-06-19:
In the small example below, Tiny Renderer returns incorrect depth values that do not match the values returned by the OpenGL renderer. The depth values also seem to vary strongly with different near and far planes.

```
import matplotlib.pyplot as plt
import numpy as np
import pybullet as p

p.connect(p.GUI)
p.loadURDF('plane.urdf')
p.loadURDF('cube_small.urdf', basePosition=[0.0, 0.0, 0.025])

width = 128
height = 128

fov = 60
aspect = width / height
near = 0.02
far = 1

view_matrix = p.computeViewMatrix([0, 0, 0.5], [0, 0, 0], [1, 0, 0])
projection_matrix = p.computeProjectionMatrixFOV(fov, aspect, near, far)

# Get depth values using the OpenGL renderer
images = p.getCameraImage(width, height, view_matrix, projection_matrix, renderer=p.ER_BULLET_HARDWARE_OPENGL)
depth_buffer_opengl = np.reshape(images[3], [width, height])
depth_opengl = far * near / (far - (far - near) * depth_buffer_opengl)

# Get depth values using Tiny renderer
images = p.getCameraImage(width, height, view_matrix, projection_matrix, renderer=p.ER_TINY_RENDERER)
depth_buffer_tiny = np.reshape(images[3], [width, height])
depth_tiny = far * near / (far - (far - near) * depth_buffer_tiny)

p.disconnect()

# Plot both images - should show depth values of 0.45 over the cube and 0.5 over the plane
plt.subplot(1, 2, 1)
plt.imshow(depth_opengl, cmap='gray', vmin=0, vmax=1)
plt.title('OpenGL Renderer')

plt.subplot(1, 2, 2)
plt.imshow(depth_tiny, cmap='gray', vmin=0, vmax=1)
plt.title('Tiny Renderer')

plt.show()
```

erwincoumans, 2017-06-19:
TinyRenderer depth values are converted to OpenGL style depth values here:
https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/TinyRendererVisualShapeConverter.cpp#L971
If you can, please share your improvements!

mbreyer, 2017-06-19:
That is why I use the same transform to get the original depth values for both OpenGL and TinyRenderer. However, the values are slightly different, for example 0.48 instead of 0.5.

erwincoumans, 2017-06-19:
It could be, because we re-compute the near/far values from the projection matrix, which introduces some error. We could test if that is the main issue here by hard-coding the real near/far values and see if that fixes the differences. See around this line:
https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/TinyRendererVisualShapeConverter.cpp#L757

erwincoumans, 2017-06-19:
>> "The depth values also seem to vary strongly with different near and far planes."

Can you give some examples of depth values differences for various near/far values?

mbreyer, 2017-06-19:
Yes, sure.
Near      Far       Depth [correct: 0.5]
0.02      1           0.480
0.02      2          0.470
0.02      10        0.462

0.1         2          0.342
0.2        2           2

0.001    2           0.499  

It seems that for small values of the near plane, results become more accurate while larger values of the near plane (or closer to the distance to objects in the scene) lead to large errors.

erwincoumans, 2017-06-19:
See my previous comment here:
https://github.com/bulletphysics/bullet3/issues/1197#issuecomment-309476682
If we use double precision for the projectionMatrix the issue would be resolved, but it would require some work to avoid breaking the existing API.

mbreyer, 2017-06-19:
Yes, that might be the problem. Are there any plans for fixing this issue in the near future? since I hoped to use the software renderer for training a RL agent on a server.

erwincoumans, 2017-06-19:
We have been doing lots of training on servers already. How is an inaccurate depth value blocking you?
Aside from that, it should be easy to fix, I just have a lot of other things on my plate, will see if I get to it.

mbreyer, 2017-06-19:
The plan is to transfer the trained controller to real world inputs in the future, so distances should match. However, it seems that choosing low values for the near plane produces results with high enough accuracy, so I should be fine. In any case, thanks for the quick reply.

erwincoumans, 2018-03-03:
Your [pull request](https://github.com/bulletphysics/bullet3/pull/1439) fixed the issue, let's close it:

[image]

closed 2018-03-03 by erwincoumans