bulletphysics/bullet3 #1199: Inverse kinematics crashes using custom UR5

cpaxton, 2017-06-20:
Here is the failure:
```
b3Error[src/BulletInverseDynamics/IDMath.cpp,73]:
size missmatch. a.cols()= 18, b.cols()= 12
Aborted (core dumped)
```
Position of EE link:
(-0.552118182182312, -0.039336759597063065, 0.5387369394302368) (-0.1695765256881714, 0.6901625990867615, 0.7034054398536682, 0.011839637532830238)

Goal position:
[-0.45, 0.2, 0.025] (0.0, 0.0, 0.0, 1.0)

Is this expected behavior when a solution cannot be computed?

erwincoumans, 2017-06-20:
We started with Samuel Buss IK library and adopted it using Bullet with various extensions/improvements with only limited testing for our purpose. We didn't expose all IK features and also didn't test robustness of the IK etc, so there are no expectations for the IK results. 

We can only figure out what's happening if you share a reproduction case (the URDF and a pybullet script that make it crash). Can you create/share it?

cpaxton, 2017-06-20:
I can share, it's part of a project that will eventually become public -- it might be easier for me just to break out a script. When I get a chance I'll do this and get back to you.

For what it's worth I don't think there's anything wrong with the URDF, PyKDL works just fine.

erwincoumans, 2017-06-27:
Yes, please share here (or private message) and I'll try to look into it (or someone else looks into it)

erwincoumans, 2017-09-12:
Without a reproduction case we cannot work on this issue. Please re-open the issue if you can provide a URDF that causes the crash. Thanks!

closed 2017-09-12 by erwincoumans

ahundt, 2017-09-12:
Here are the files zipped up [ur5_robotiq_85.zip](https://github.com/bulletphysics/bullet3/files/1297480/ur5_robotiq_85.zip)

arpit15, 2017-09-14:
Facing similar issues with loading of baxter robot. 
[testingforbaxterloading.zip](https://github.com/bulletphysics/bullet3/files/1304360/testingforbaxterloading.zip)

erwincoumans, 2017-09-14:
The person who will look into is is busy with some paper deadline at the moment. We will look at it soon.

arpit15, 2017-09-14:
@erwincoumans Thank you for the prompt reply. Good luck for the deadline.