bulletphysics/bullet3 #1207: pip install pybullet on Anaconda 3.5, maxOS sierra 10.12.5

shaneshixiang, 2017-06-24:
I encountered the following error: 

src/LinearMath/btQuickprof.cpp:704:10: error: thread-local storage is not supported for the current target
                    static __thread unsigned int sThreadIndex = kNullIndex;
                           ^
    5 warnings and 1 error generated.
    error: command 'clang' failed with exit status 1

I also tried with env CC=clang++ CXX=clang++ as suggested here (https://github.com/scikit-hep/root_numpy/issues/209); however, received the same error. It may be due to specific local setups but any help is appreciated. Thank you!

erwincoumans, 2017-06-24:
Odd, it works just fine here. What version of Xcode/clang/gcc do you use?

Can you upgrade to Anaconda 3.6? There is a wheel that is precompiled.

shaneshixiang, 2017-06-25:
xcodebuild -version

Xcode 8.3.3

Build version 8E3004b

clang --version

Apple LLVM version 8.1.0 (clang-802.0.42)

Target: x86_64-apple-darwin16.6.0

gcc --version

Apple LLVM version 8.1.0 (clang-802.0.42)

Target: x86_64-apple-darwin16.6.0

Actually I have been using Anaconda 4.2.9. With python 3.5 environment.

Thank you!

erwincoumans, 2017-06-25:
If you use Python 3.6 environment there is a wheel for pybullet, not for Python 3.5.

The issue is that your compiler doesn't seem to support __thread, while mine does, and the version is the same???

If you copy the contents between ------------------- in main.cpp, can you try compiling it?

gcc main.cpp
clang main.cpp
--------------------------------
__thread int t;
int main(int argc, char* argv[])
{
}
--------------------------------

erwincoumans, 2017-06-25:
If you figure out why your machine doesn't support __thread (while it should) and how to fix it, please share it.

mvandepanne, 2017-10-16:
I had the same issue on MacOS. Changing to Python 3.6 solved it for me.

sungsulim, 2018-04-06:
I have the exact same installation issue on MacOS, and I'm using python2.7, with Anaconda.
I tried:
$CFLAGS="-DB3_NO_PYTHON_FRAMEWORK" pip install pybullet -v

$ python --version
Python 2.7.14 :: Anaconda, Inc.

$ conda -V
conda 4.4.10

$ gcc --version
Apple LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin17.4.0
Thread model: posix

$ clang --version
Apple LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin17.4.0
Thread model: posix

Any help would be appreciated.
Thanks!

sungsulim, 2018-04-07:
I got it working by building from source instead. 
 https://github.com/bulletphysics/bullet3/issues/830

erwincoumans, 2018-12-23:
Closing the issue, PyBullet should work with the regular version of Python 2.7 and 3.5/3.6 through Homebrew (brew install). Anaconda versions work for me, but others have issues. It must be some configuration issue that we cannot dig into. Let's close it.

closed 2018-12-23 by erwincoumans