bulletphysics/bullet3 #1209: Compiling bullet3 with python if your python setup is messed up

benelot, 2017-06-24:
Simply put, my cmake does not find python properly on its own. If you encounter this problem, just find the python interpreter which shows the appropriate python version on your console. For me this was 'python'. For you this might be python2.7 or similar. Then edit the following to the build_cmake_pybullet_double.sh file:

	cmake -DBUILD_PYBULLET=ON -DBUILD_PYBULLET_NUMPY=OFF -DUSE_DOUBLE_PRECISION=ON -DCMAKE_BUILD_TYPE=Release  ..

to

	cmake -DBUILD_PYBULLET=ON -DBUILD_PYBULLET_NUMPY=OFF -DUSE_DOUBLE_PRECISION=ON -DCMAKE_BUILD_TYPE=Release \
	-DPYTHON_INCLUDE_DIR=$(python -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())")  \
	-DPYTHON_LIBRARY=$(python -c "import distutils.sysconfig as sysconfig; print(sysconfig.get_config_var('LIBDIR'))") ..

Replace the 'python' in '$(python -c' in the above lines with the interpreter name you found. This will the automatically resolve for the correct python directories and submit them to cmake. Best of luck!

cprogrammer1994, 2017-06-24:
Would not be better to provide wheels on PyPI?

cprogrammer1994, 2017-06-24:
I better say "easier" not "better" 😄

erwincoumans, 2017-06-24:
You should be able to use pip install pybullet indeed, that is recommended over using cmake etc.

https://pypi.python.org/pypi/pybullet

No need for wheels, pip should just install from source.

I upload some Mac OSX and Windows wheels sometimes. Would be nice to have an automatic way to generate wheels for most popular versions of python on Mac and Windows.

erwincoumans, 2017-06-24:
Benjamin, should we replace the line in build_cmake_pybullet_double.sh  file with your one? 

I would assume 'python' to be the default name in most cases.

benelot, 2017-06-24:
I can not yet say if this works in all cases, but in mine, it fixed the issue. We can change it and see how many other people's build system we break ;-) No, in fact it should work in general as it returns the correct path for that interpreter. It remains in question why the cmake search fails...

erwincoumans, 2017-06-24:
cmake fails to find packages so many times, especially on Windows. I spend a lot of time digging into some cmake files to see how it finds a particular package: cmake doesn't make it easy at all to find packages automagically.

pip install pybullet is preferred. On Windows, premake works better than cmake.

benelot, 2017-06-24:
Well, then it is maybe best if you go ahead and add it to the file, it is more fail-safe.

benelot, 2017-06-25:
I made a pull-request for you in #1211.

erwincoumans, 2017-09-24:
We got pip now, use pip install pybullet to install pybullet, don't waste your (and ours) time trying to tell cmake where to find python.

closed 2017-09-24 by erwincoumans

benelot, 2017-09-25:
Sounds great! I will use the pip version in the future.