bulletphysics/bullet3 #1212: Docker help

alee156, 2017-06-28:
Do you know of any Docker containers using bullet3 that use the shared memory connect option for pybullet? I'm trying to set up a Docker container that I can access the gui.

Thanks!
Albert

benelot, 2017-08-26:
There are no docker containers for bullet3 that I know of. Are you sure you can connect into a docker container using shared memory? In fact the probably do not even share memory. Maybe try the TCP option, that would be my best guess.

jtoy, 2018-02-09:
has anyone successfully built bullet in Docker, I'm getting a similar error related to shared memory  building in ubuntu 16.04:

obj/x64/Release/App_PhysicsServer_SharedMemory/b3PluginManager.o:b3PluginManager.cpp:(.text+0x255f): more undefined references to `dlsym' follow
obj/x64/Release/App_PhysicsServer_SharedMemory/b3PluginManager.o: In function `b3PluginManager::loadPlugin(char const*, char const*)':
b3PluginManager.cpp:(.text+0x25f8): undefined reference to `dlclose'
b3PluginManager.cpp:(.text+0x2750): undefined reference to `dlclose'
b3PluginManager.cpp:(.text+0x2aa2): undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
App_PhysicsServer_SharedMemory.make:152: recipe for target '../../bin/App_PhysicsServer_SharedMemory_gmake_x64_release' failed
make[1]: *** [../../bin/App_PhysicsServer_SharedMemory_gmake_x64_release] Error 1
Makefile:40: recipe for target 'App_PhysicsServer_SharedMemory' failed
make: *** [App_PhysicsServer_SharedMemory] Error 2

cprogrammer1994, 2018-02-09:
Try to add `-ldl` option to the linker.

erwincoumans, 2018-12-23:
Closing the issue, we don't support or deal with Docker.

closed 2018-12-23 by erwincoumans