bulletphysics/bullet3 #1213: btCreateCompoundFromGimpactShape leads to memory leak

Wugand, 2017-06-29:
The function btCreateCompoundFromGimpactShape is allocating much memory internally. (Which goes somewhere into the btCompoundShape). But this memory is not deallocated, when deleting the btCompundShape.

Analysis said the unreferenced heap space is allocated by:
`btBU_Simplex1to4* tet = new btBU_Simplex1to4(v0,v1,v2,rayTo);` (Line 77)
*It seams to me that this object is stored somewhere in "m_children" of btCompoundShape

Why is this memory doesn't get deallocated automatically?
Is there some way how I can deallocate this remains? 

Thanks!

(defined in: bullet3/src/BulletCollision/Gimpact/btCompoundFromGimpact.h)
https://github.com/bulletphysics/bullet3/blob/79d24db60f2c916ffcdaf8b074be622e2e5bc609/src/BulletCollision/Gimpact/btCompoundFromGimpact.h

erwincoumans, 2017-07-02:
Needs fixing indeed, pull request?

erwincoumans, 2018-12-23:
Thanks! Closing old issue.

closed 2018-12-23 by erwincoumans