bulletphysics/bullet3 #1215: pip install pybullet fails on fresh ubuntu machine

matpalm, 2017-06-30:
setting up a completely fresh `Ubuntu 16.04.2 LTS` machine and `pip install pybullet` fails with an X11 dependency..

```
In file included from examples/OpenGLWindow/X11OpenGLWindow.cpp:13:0:
examples/ThirdPartyLibs/Glew/CustomGL/glxew.h:97:22: fatal error: X11/Xlib.h: No such file or directory
compilation terminated.
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

as expected it installs fine after `sudo apt-get install libx11-dev` but thought this would be good to somehow document somehow since it's the sort of thing that would scare a new user. it's a rare test case of a completely fresh machine!

erwincoumans, 2017-07-01:
Good point. We could enable some define to avoid the need for X11 actually (that's what premake does). thanks for the feedback!

erwincoumans, 2017-07-02:
x11 headers are not needed anymore, with this commit:https://github.com/bulletphysics/bullet3/commit/e6f7eb486a1da9a5f589f56b3aec5126ea6b8a90

closed 2017-07-02 by erwincoumans