bulletphysics/bullet3 #1226: [feature] Python Wrapper -> cffi

realitix, 2017-07-13:
Hello @erwincoumans,

First, I would like to thank you for this amazing piece of work you did (almost alone!).
I saw that recently you started to use Python (a very good language btw). I'm a fan of Python so I'm enjoying this direction.

From what I saw, you are directly using the CPython C API to create your extension module. This way offers you a great liberty but at a great cost, ultimate dependency with CPython. As you may know, Python can run with others interpreters than CPython, the most famous one is PyPy and offers better performance than CPython under certain circumstances.

The problem is that PyPy doesn't natively support the CPython C API (which is deeply linked with CPython internal code) and suffers of bad performance when using Python extension written for CPython. Fortunately, there is a solution which is very good: cffi.

CFFI is a Python module that allows you to easily write Python extension with full compatibility over CPython and PyPy. Moreover, it's a lot easier to write an extension with cffi than with CPython C API.

I would like to know if you ever considerate cffi as a solution or if maybe you don't know it.
I look forward to see your though about it.

Thanks in advance for your answer,
Sincerely,
Jean-Sébastien

erwincoumans, 2017-07-14:
I was not aware of this issue. Can CFFI create automatic Python bindings, given a C - header file?

If so, please show it to me, create a binding using this C header file:
bullet/examples/SharedMemory/PhysicsClientC_API.h

In addition, or alternatively, can CFFI bind a C++ header? If so, please try binding this:
bullet3/examples/RobotSimulator/b3RobotSimulatorClientAPI.h

Thanks!
Erwin

realitix, 2017-07-16:
Hello @erwincoumans,

CFFI can create a binding automatically given a C header file. Nevertheless, it won't run the preprocessor. It's not really a problem, usually, I execute the preprocessor with `cpp` command and I feed the cffi's `cdef` function with it.

CFFI can't parse C++ header. Is your current Python extension using the C++ header file ?

Let me just some time, I will show you how it works with a pull request. I just have to figure out how all is working.

Thanks,
Jean-Sébastien

erwincoumans, 2017-07-16:
Pybulley manually binds the C API, not the C++ API. Having an example of
both would be helpful.

erwincoumans, 2018-12-23:
We stick with PyBullet for Bullet. Let's close this issue.

closed 2018-12-23 by erwincoumans