bulletphysics/bullet3 #1227: Collision penetration resolve

ongamex, 2017-07-13:
Hello guys. I'm pretty new to Bullet Physics and physics engine as a whole.

I'm trying to create a physics for a 3D platformer. Where the player is represented by a dynamics in rotatable capsule, Everything is good so far, but the problem I have is when the player falls down on a slope:

[image]

Moving the player outside of the ground the do the trick, but I do not know how to achieve it.
I chose to use a dynamic instead of a kinematic one because It look easier to implement and I'm in a research phase and this is what I've started with, additionally I want my object to be able to push and to be pushable by other object.

Additionally all tips on the subject are welcome as I do not know how to approach this problem.

Regards,
me :)

BluePrintRandom, 2017-07-13:
in bge python

if state=="idle" and GroundRay.positive:
(tab)own.localLinearVelocity*=.5
(tab)own.applyForce([0,0,own.mass*gravity], 0)

If on ground, cancel gravity and apply homemade friction.

ongamex, 2017-07-13:
Thank you for the fast response.

What I do is I cast a few rays from the bottom of the capsule in order to determine if the player is grounded. If so I indeed disable the velocity, However the object could still penetrate the ground, if it is moving a bit faster.

BluePrintRandom, 2017-07-13:
See if CCD can be turned on in your platform
(I use bge python which lacks CCD at the moment )

I personally raycast based on LinV and a multiplier.

The faster the agent  moves the longer the ray becomes, and if it
penetrates a collider, I move the object along the vector to the hitpoint,
which is simple for a sphere, but can get tricky for convex hulls etc.
(like deciding to raycast from leading edges based on linv)

ongamex, 2017-07-14:
Thanks. I'll try it tonight. My problem is not tunneling but I guess this might help!

ongamex, 2017-07-17:
Unfortunately this did not help.
I thought that this is a different but now that I think about it is the same thing:

If I have the player's capsule and a ground plane, and I drop above the plane, when the capsule land again the player sinks for a bit and for a few frames slowly emerges on top of the plane? How can I resolve this?

I'm using one substep in stepSimulation, because I want to catch every intersection with the environment in order to make the rigid body's behave more like a character.

romieval, 2017-07-28:
If I correctly remember Bullet's character controllers have dedicated logical code to prevent the fall.

erwincoumans, 2018-12-23:
Character control is a topic on its own, and people develop their own solutions. We don't have volunteers working on this, so closing the issue.

closed 2018-12-23 by erwincoumans