bulletphysics/bullet3 #1233: Why this snippet cause Access Violation

cprogrammer1994, 2017-07-17:
Hello,

I have trouble creating a `btMultiBody` with a base collider and a single link fixed to it.

I compile with a single command using g++

The code gets an Access Violation when the first `stepSimulation` is called.

- With no base collider + a single fixed link everything works fine
- With a base collider + no fixed links everything works fine

Please help me understand this, or guide me to an example where I can find similar code.

```s
g++ -m32 -mstackrealign -I bullet -D BT_USE_DOUBLE_PRECISION -D BT_NO_PROFILE main.cpp <all the cpp files from the bullet source>
```

Here is my main.cpp

```c++
#include <cstdio>
#include <btBulletDynamicsCommon.h>

#include "BulletDynamics/Featherstone/btMultiBody.h"
#include "BulletDynamics/Featherstone/btMultiBodyConstraintSolver.h"
#include "BulletDynamics/Featherstone/btMultiBodyDynamicsWorld.h"
#include "BulletDynamics/Featherstone/btMultiBodyLinkCollider.h"
#include "BulletDynamics/Featherstone/btMultiBodyLink.h"
#include "BulletDynamics/Featherstone/btMultiBodyJointLimitConstraint.h"
#include "BulletDynamics/Featherstone/btMultiBodyJointMotor.h"
#include "BulletDynamics/Featherstone/btMultiBodyPoint2Point.h"
#include "BulletDynamics/Featherstone/btMultiBodyFixedConstraint.h"
#include "BulletDynamics/Featherstone/btMultiBodySliderConstraint.h"

int main() {
	btCollisionConfiguration * collision_configuration = new btDefaultCollisionConfiguration();
	btCollisionDispatcher * dispatcher = new btCollisionDispatcher(collision_configuration);
	btDbvtBroadphase * broadphase = new btDbvtBroadphase();
	btMultiBodyConstraintSolver * solver = new btMultiBodyConstraintSolver();
	btMultiBodyDynamicsWorld * world = new btMultiBodyDynamicsWorld(
		dispatcher,
		broadphase,
		solver,
		collision_configuration
	);

	world->setGravity(btVector3(0, 0, -9.8));

	btVector3 inertia = btVector3(2.0 / 3.0, 2.0 / 3.0, 2.0 / 3.0);
	btMultiBody * multi_body = new btMultiBody(1, 1.0, inertia, false, false, false);

	btTransform transform;
	transform.setIdentity();
	multi_body->setBaseWorldTransform(transform);

	{
		btBoxShape * shape = new btBoxShape(btVector3(1.0, 1.0, 1.0));
		btMultiBodyLinkCollider * collider = new btMultiBodyLinkCollider(multi_body, -1);
		collider->setCollisionShape(shape);

		multi_body->setBaseCollider(collider);
	}

	{
		btBoxShape * shape = new btBoxShape(btVector3(1.0, 1.0, 1.0));
		btMultiBodyLinkCollider * collider = new btMultiBodyLinkCollider(multi_body, 0);
		collider->setCollisionShape(shape);

		multi_body->getLink(0).m_collider = collider;
	}

	multi_body->setupFixed(
		0,
		1.0,
		inertia,
		-1,
		btQuaternion(0.0, 0.0, 0.0, 1.0),
		btVector3(0.0, 0.0, 0.0),
		btVector3(0.0, 0.0, 0.0),
		false
	);

	multi_body->finalizeMultiDof();
	world->addMultiBody(multi_body);

	btScalar delta_time = 1.0 / 60.0;
	world->stepSimulation(delta_time, 0, delta_time);
	world->stepSimulation(delta_time, 0, delta_time);
	world->stepSimulation(delta_time, 0, delta_time);
	printf("Done!");
}
```

Any help or hint would be appreciated,

Thank you!

cprogrammer1994, 2017-07-17:
I insatlled visual studio and I got a btAssert called. Here is my call stack:

[image]

erwincoumans, 2019-12-18:
Closing out-of-date issues (cannot replicate, PyBullet heavily uses btMultiBody and no such report)

closed 2019-12-18 by erwincoumans