bulletphysics/bullet3 #1234: Steps for running gym examples?

ahundt, 2017-07-18:
I saw the greatly expanded gym examples the selection looks nice!  I checked the pybullet docs and I tried running them, but the path doesn't seem to work as-is, and it seems `spaces` is missing.

benelot, 2017-07-18:
Which examples do you mean? The ones that are in in the folder gym so far?

erwincoumans, 2017-07-18:
For the current gym examples, you need an up-to-date versions of pybullet, Tensorflow, OpenAI Gym and OpenAI Baselines (sudo pip install pybullet, tensorflow, gym, baselines).
Then open a terminal windows, change directory to Bullet/examples/pybullet/gym and you should be able to run:
python train_pybullet_racecar.py
wait until it is trained and terminates this can take some hours(s), then run
python enjoy_pybullet_racecar.py

What operating system did you use? What exact error(s) do you get (copy paste them)

ahundt, 2017-07-19:
What about having setup.py another flag in setup.py (or another setup.py) for those examples? That way it could incorporate a `requirements.txt` file and install the deps.

I'll give this another try and post more detailed errors (or success) in a bit. Sorry I didn't post more detail because I had been on a plane when I tried it and suspected I was missing dependencies, the errors were long gone when I next had net access.

ahundt, 2017-07-19:
Here are the specific errors, I made sure all the packages are installed.

```
athundt at Andrews-2013-MacBook-Pro-2 in ~/src/bullet3/examples/pybullet/gym on master!
± python enjoy_kuka_grasping.py
Traceback (most recent call last):
  File "enjoy_kuka_grasping.py", line 2, in <module>
    from envs.bullet.kukaGymEnv import KukaGymEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/__init__.py", line 1, in <module>
    from envs.bullet.cartpole_bullet import CartPoleBulletEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/cartpole_bullet.py", line 9, in <module>
    from gym import spaces
ImportError: cannot import name spaces

```

```
athundt at Andrews-2013-MacBook-Pro-2 in ~/src/bullet3/examples/pybullet/gym on master!
± python train_pybullet_racecar.py
Traceback (most recent call last):
  File "train_pybullet_racecar.py", line 2, in <module>
    from envs.bullet.racecarGymEnv import RacecarGymEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/__init__.py", line 1, in <module>
    from envs.bullet.cartpole_bullet import CartPoleBulletEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/cartpole_bullet.py", line 9, in <module>
    from gym import spaces
ImportError: cannot import name spaces
```

```
± pip install gym
Requirement already satisfied: gym in /usr/local/lib/python2.7/site-packages
Requirement already satisfied: pyglet>=1.2.0 in /usr/local/lib/python2.7/site-packages (from gym)
Requirement already satisfied: numpy>=1.10.4 in /usr/local/lib/python2.7/site-packages (from gym)
Requirement already satisfied: six in /usr/local/lib/python2.7/site-packages (from gym)
Requirement already satisfied: requests>=2.0 in /usr/local/lib/python2.7/site-packages (from gym)
```

erwincoumans, 2017-07-19:
Before running the other environments, please run 

python train_pybullet_racecar.py
wait until it is trained and terminates this can take some hours(s), then run
python enjoy_pybullet_racecar.py

ahundt, 2017-07-19:
so embarrassing! I skimmed too quickly. sorry

erwincoumans, 2017-07-19:
Did you install openai baselines?

ahundt, 2017-07-19:
Similar error:

```
± python train_pybullet_racecar.py
Traceback (most recent call last):
  File "train_pybullet_racecar.py", line 2, in <module>
    from envs.bullet.racecarGymEnv import RacecarGymEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/__init__.py", line 1, in <module>
    from envs.bullet.cartpole_bullet import CartPoleBulletEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/cartpole_bullet.py", line 9, in <module>
    from gym import spaces
ImportError: cannot import name spaces
```
here are all my packages, gym is in there, as is baselines.
```
± pip freeze
alabaster==0.7.10
apipkg==1.4
appdirs==1.4.3
appnope==0.1.0
astroid==1.4.9
autopep8==1.3.1
azure==1.0.3
azure-common==1.1.6
azure-mgmt==0.20.2
azure-mgmt-common==0.20.0
azure-mgmt-compute==0.20.1
azure-mgmt-network==0.20.1
azure-mgmt-nspkg==2.0.0
azure-mgmt-resource==0.20.1
azure-mgmt-storage==0.20.0
azure-nspkg==2.0.0
azure-servicebus==0.20.1
azure-servicemanagement-legacy==0.20.2
azure-storage==0.20.3
Babel==2.4.0
backports.functools-lru-cache==1.3
backports.ssl-match-hostname==3.5.0.1
backports.weakref==1.0rc1
baselines==0.1.4
bleach==1.5.0
bloom==0.5.26
breathe==4.1.0
caffe2keras==0.0.1
catkin-pkg==0.3.1
catkin-tools==0.4.4
click==6.7
configparser==3.5.0
coverage==4.0.3
cycler==0.10.0
Cython==0.25.2
decorator==4.0.11
dill==0.2.7.1
docker-py==1.10.6
docker-pycreds==0.2.1
docopt==0.6.2
docutils==0.13.1
ecdsa==0.13
empy==3.3.2
enum34==1.1.6
execnet==1.4.1
Flask==0.12
funcsigs==1.0.2
functools32==3.2.3.post2
futures==3.1.1
gevent==1.2.1
gnureadline==6.3.3
greenlet==0.4.12
gtabview==0.8
gym==0.9.2
h5py==2.7.0
html5lib==0.9999999
imageio==2.1.2
imagesize==0.7.1
ipaddress==1.0.18
ipython==4.1.2
ipython-genutils==0.1.0
isort==4.2.5
itsdangerous==0.24
Jinja2==2.9.6
joblib==0.11
Keras==2.0.5
keras-contrib==1.2.1
lazy-object-proxy==1.2.2
llvmlite==0.18.0
lmdb==0.89
Markdown==2.2.0
MarkupSafe==1.0
matplotlib==2.0.0
mc-cython==0.0.0
mccabe==0.6.1
mercurial==4.2.1
mock==2.0.0
moviepy==0.2.3.2
netifaces==0.10.4
networkx==1.11
nose==1.3.7
numba==0.34.0.dev0+410.g754ab0423
numpy==1.13.0
olefile==0.44
osrf-pycommon==0.1.2
packaging==16.8
pandas==0.20.1
paramiko==1.16.0
path.py==8.1.2
pbr==3.1.0
pep8==1.7.0
pexpect==4.0.1
pickleshare==0.6
Pillow==4.1.0
pkgconfig==1.2.2
progressbar2==3.30.2
protobuf==3.3.0
psutil==3.4.2
ptk==1.3.2
ptvsd==2.2.0
ptyprocess==0.5.1
py==1.4.33
PyBindGen==0.17.0
pybullet==1.2.2
pycocotools==2.0
pycodestyle==2.3.1
pycrypto==2.6.1
pycurl==7.43.0
pydot==1.0.2
pyglet==1.2.4
Pygments==2.2.0
pygraphviz==1.3.1
pylint==1.6.5
pymongo==3.4.0
PyOpenGL==3.1.0
pyparsing==2.2.0
pytest==3.0.7
pytest-cov==2.4.0
pytest-xdist==1.15.0
python-dateutil==2.6.0
python-utils==2.1.0
pytz==2017.2
PyWavelets==0.5.2
PyYAML==3.12
pyzmq==16.0.2
requests==2.14.2
rosdep==0.11.5
rosdistro==0.6.2
rosinstall==0.7.8
rosinstall-generator==0.1.13
rospkg==1.1.1
sacred==0.6.10
sacredboard==0.1.2
scikit-image==0.13.0
scikit-learn==0.18.1
scipy==0.19.0
simplegeneric==0.8.1
singledispatch==3.4.0.3
six==1.10.0
sklearn==0.0
snowballstemmer==1.2.1
Sphinx==1.6.1
sphinx-rtd-theme==0.1.9
sphinxcontrib-websupport==1.0.1
subprocess32==3.2.7
TBB==0.1
tensorflow==1.2.0
tensorlayer==1.4.3
Theano==0.9.0
tqdm==4.11.2
traitlets==4.2.1
trollius==2.1
typing==3.6.1
vcstools==0.1.39
virtualenv==15.1.0
websocket-client==0.40.0
Werkzeug==0.12.2
wrapt==1.10.8
wstool==0.1.13
xonsh-docker-tabcomplete==0.1.4
xonsh-vox-tabcomplete==0.4
xontrib-avox==0.6
xontrib-z==0.3
zmq==0.0.0
```

erwincoumans, 2017-07-19:
Python version?

erwincoumans, 2017-07-19:
Odd, you gym installation seems broken, not a pybullet issue.
If you run python and type "from gym import spaces", do you get an error?
If so, please report it in the OpenAI gym github tracker (or other OpenAI bug report mechanism)

ahundt, 2017-07-19:
I'm on python 2.7.13. I had a stray folder causing that spaces problem, so I've run into something else. Perhaps it is also a problem with baselines?

```
athundt at Andrews-2013-MacBook-Pro-2 in ~/src/bullet3/examples/pybullet/gym on master!
± python train_pybullet_racecar.py
pybullet build time: Jul 17 2017 18:59:54
Couldn't import dot_parser, loading of dot files will not be possible.
Traceback (most recent call last):
  File "train_pybullet_racecar.py", line 4, in <module>
    from baselines import deepq
  File "/usr/local/lib/python2.7/site-packages/baselines/deepq/__init__.py", line 2, in <module>
    from baselines.deepq.build_graph import build_act, build_train  # noqa
  File "/usr/local/lib/python2.7/site-packages/baselines/deepq/build_graph.py", line 71, in <module>
    import baselines.common.tf_util as U
  File "/usr/local/lib/python2.7/site-packages/baselines/common/tf_util.py", line 3, in <module>
    import builtins
ImportError: No module named builtins
```

train_kuka_grasping.py
```
± python train_kuka_grasping.py
pybullet build time: Jul 17 2017 18:59:54
Couldn't import dot_parser, loading of dot files will not be possible.
Traceback (most recent call last):
  File "train_kuka_grasping.py", line 4, in <module>
    from baselines import deepq
  File "/usr/local/lib/python2.7/site-packages/baselines/deepq/__init__.py", line 2, in <module>
    from baselines.deepq.build_graph import build_act, build_train  # noqa
  File "/usr/local/lib/python2.7/site-packages/baselines/deepq/build_graph.py", line 71, in <module>
    import baselines.common.tf_util as U
  File "/usr/local/lib/python2.7/site-packages/baselines/common/tf_util.py", line 3, in <module>
    import builtins
ImportError: No module named builtins
```

ahundt, 2017-07-19:
I also reported this at https://github.com/openai/baselines/issues/57

ahundt, 2017-07-20:
It seems baselines only supports python2

ahundt, 2017-07-20:
I tried running the version from pip3 in python3 and got the following:

```
± python3 train_kuka_cam_grasping.py
Traceback (most recent call last):
  File "train_kuka_cam_grasping.py", line 2, in <module>
    from envs.bullet.kukaCamGymEnv import KukaCamGymEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/__init__.py", line 1, in <module>
    from envs.bullet.cartpole_bullet import CartPoleBulletEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/cartpole_bullet.py", line 14, in <module>
    import pybullet as p
ImportError: dlopen(/usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so, 2): Symbol not found: _PyInt_FromLong
  Referenced from: /usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so
  Expected in: flat namespace
 in /usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so
```

erwincoumans, 2017-07-20:
Please stop using the other tests (kuka etc), only report python train_pybullet_racecar.py for now.

Baselines works fine with Python 3.6/3.5 here. On Mac OSX when using Python 3 (non-system), you need to use the following (and sudo pip uninstall pybullet first, if needed):

CFLAGS="-DB3_NO_PYTHON_FRAMEWORK" pip install pybullet

Please let me know the result of using python train_pybullet_racecar.py

ahundt, 2017-07-20:
Here is my session (I had to change pip to pip3 in my case):

```
± CFLAGS="-DB3_NO_PYTHON_FRAMEWORK" pip3 install pybullet
Collecting pybullet
Installing collected packages: pybullet
Successfully installed pybullet-1.2.2

± cd examples/pybullet/gym

± python3 train_pybullet_racecar.py
Traceback (most recent call last):
  File "train_pybullet_racecar.py", line 2, in <module>
    from envs.bullet.racecarGymEnv import RacecarGymEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/__init__.py", line 1, in <module>
    from envs.bullet.cartpole_bullet import CartPoleBulletEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/cartpole_bullet.py", line 14, in <module>
    import pybullet as p
ImportError: dlopen(/usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so, 2): Symbol not found: _PyInt_FromLong
  Referenced from: /usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so
  Expected in: flat namespace
 in /usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so
```

ahundt, 2017-08-01:
well pip3 install pybullet seems to work ok now, I wonder why setup.py builds but is missing that symbol at runtime.

ahundt, 2017-08-01:
actually I'm still having the same problem with the train script:

```
athundt at Andrews-2013-MacBook-Pro-2 in ~/src/bullet3/examples/pybullet/gym on master!
± python3 train_pybullet_racecar.py
Traceback (most recent call last):
  File "train_pybullet_racecar.py", line 2, in <module>
    from envs.bullet.racecarGymEnv import RacecarGymEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/__init__.py", line 1, in <module>
    from envs.bullet.cartpole_bullet import CartPoleBulletEnv
  File "/Users/athundt/source/bullet3/examples/pybullet/gym/envs/bullet/cartpole_bullet.py", line 14, in <module>
    import pybullet as p
ImportError: dlopen(/usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so, 2): Symbol not found: _PyInt_FromLong
  Referenced from: /usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so
  Expected in: flat namespace
 in /usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so

athundt at Andrews-2013-MacBook-Pro-2 in ~/src/bullet3/examples/pybullet/gym on master!
± python train_pybullet_racecar.py
pybullet build time: Aug  1 2017 11:21:09
Couldn't import dot_parser, loading of dot files will not be possible.
Traceback (most recent call last):
  File "train_pybullet_racecar.py", line 4, in <module>
    from baselines import deepq
  File "/usr/local/lib/python2.7/site-packages/baselines/deepq/__init__.py", line 2, in <module>
    from baselines.deepq.build_graph import build_act, build_train  # noqa
  File "/usr/local/lib/python2.7/site-packages/baselines/deepq/build_graph.py", line 71, in <module>
    import baselines.common.tf_util as U
  File "/usr/local/lib/python2.7/site-packages/baselines/common/tf_util.py", line 3, in <module>
    import builtins
ImportError: No module named builtins
```

ahundt, 2017-08-04:
Sorry to press this but what are the next steps here? I usually try to fix problems myself and submit pull requests but I'm not very familiar with the Python -> C++ interface and how linking is done in this situation. Perhaps you could point me to a resource I could use to solve the problem myself?

erwincoumans, 2017-08-09:
What version of Python3 are you using? I am using Anaconda3 and it works just fine.
(please stop reporting results using Python2)

erwincoumans-macbookpro:humanoid erwincoumans$ which python
/Users/erwincoumans/anaconda3/bin/python
erwincoumans-macbookpro:humanoid erwincoumans$ /Users/erwincoumans/anaconda3/bin/python --version
Python 3.6.0 :: Anaconda custom (x86_64)

erwincoumans-macbookpro:gym erwincoumans$ which python
/Users/erwincoumans/anaconda3/bin/python
erwincoumans-macbookpro:gym erwincoumans$ which pip
/Users/erwincoumans/anaconda3/bin/pip

erwincoumans, 2017-08-09:
When pybullet is compiled against Python 3.x, there should not be any PyInt_FromLong.
It seems your pybullet for Python 3.x is not properly compiled. See for example in Bullet/examples/pybullet/pybullet.c

#if PY_MAJOR_VERSION >= 3
#define PyInt_FromLong PyLong_FromLong
#define PyString_FromString PyBytes_FromString
#endif

Can you uninstall pybullet and re-build/re-install it again using:
CFLAGS="-DB3_NO_PYTHON_FRAMEWORK" sudo python3 setup.py install

Or even better uninstall your python3 and re-install the Anaconda3 version of Python 3.6 and re-install pybullet using CFLAGS="-DB3_NO_PYTHON_FRAMEWORK" sudo pip install pybullet
(assuming python and pip are the Anaconda3 versions, not the system version in /usr/bin/python etc) The Anaconda installer should take care of this.

ahundt, 2017-08-09:
Would you mind mentioning which install process you follow for day to day development so I can try my hand at fixing the linking problem on my platform in the best supported manner?

I checked the README, coding style and quick setup guides first. Hopefully I didn't miss it.

> (please stop reporting results using Python2)

Sorry, I didn't realize py2 is unsupported, I also have homebrew's python3:

```
○ which python3
/usr/local/bin/python3
○ brew info python3
python3: stable 3.6.2 (bottled), HEAD
Interpreted, interactive, object-oriented programming language
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/python3.rb
```

> Or even better uninstall your python3 and re-install the Anaconda3 version of Python 3.6 and re-install pybullet using CFLAGS="-DB3_NO_PYTHON_FRAMEWORK"

I'll see if I can learn how to use virtualenv to do this temporarily, upgrading my actual code base would take too long. :-) Is the reason for anaconda3 because it is the version you use?

I will try each approach shortly. Thanks!

erwincoumans, 2017-08-09:
>>  Is the reason for anaconda3 because it is the version you use?

Yes, because there are a gazillion permutations of versions of Python, installer, build system, and I don't have cycles to support them all.

erwincoumans, 2017-08-09:
I just installed homebrew python3 on my Mac, and 'import pybullet' works just fine. 

brew install python3
pip3 install pybullet
pip3 install numpy
python3
>>>import pybullet as p
pybullet build time: Jul 20 2017 16:05:39

What do you get if you run python3 and import pybullet as p?

ahundt, 2017-08-09:
>  'import pybullet' works just fine.

Apparently my computer is being evil then... :-(

```bash
± pip3 uninstall pybullet
Uninstalling pybullet-1.2.2:
  /usr/local/lib/python3.6/site-packages/pybullet-1.2.2.dist-info/DESCRIPTION.rst
  /usr/local/lib/python3.6/site-packages/pybullet-1.2.2.dist-info/INSTALLER
  /usr/local/lib/python3.6/site-packages/pybullet-1.2.2.dist-info/METADATA
  /usr/local/lib/python3.6/site-packages/pybullet-1.2.2.dist-info/RECORD
  /usr/local/lib/python3.6/site-packages/pybullet-1.2.2.dist-info/WHEEL
  /usr/local/lib/python3.6/site-packages/pybullet-1.2.2.dist-info/metadata.json
  /usr/local/lib/python3.6/site-packages/pybullet-1.2.2.dist-info/top_level.txt
  /usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so
Proceed (y/n)? y
  Successfully uninstalled pybullet-1.2.2

athundt at Andrews-2013-MacBook-Pro-2 in ~/src/bullet3/examples/pybullet/examples on master!
± ls /usr/local/lib/python3.6/site-packages/pybullet*
zsh: no matches found: /usr/local/lib/python3.6/site-packages/pybullet*

athundt at Andrews-2013-MacBook-Pro-2 in ~/src/bullet3/examples/pybullet/examples on master!
± pip3 install pybullet
Collecting pybullet
Installing collected packages: pybullet
Successfully installed pybullet-1.2.2

± ls /usr/local/lib/python3.6/site-packages/pybullet*
/usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so

/usr/local/lib/python3.6/site-packages/pybullet-1.2.2.dist-info:
DESCRIPTION.rst METADATA        WHEEL           top_level.txt
INSTALLER       RECORD          metadata.json

athundt at Andrews-2013-MacBook-Pro-2 in ~/src/bullet3/examples/pybullet/examples on master!
± python3 -c 'import pybullet as p'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: dlopen(/usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so, 2): Symbol not found: _PyInt_FromLong
  Referenced from: /usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so
  Expected in: flat namespace
 in /usr/local/lib/python3.6/site-packages/pybullet.cpython-36m-darwin.so
athundt at Andrews-2013-MacBook-Pro-2 in ~/src/bullet3/examples/pybullet/examples on master!

```

erwincoumans, 2017-08-09:
Odd. What is the version of Mac OSX and version of gcc?

ahundt, 2017-08-09:
sorry just edited above with most of the information

also OS X 10.11.6 (15G1510)

ahundt, 2017-08-09:
My default python i.e. `which python` is python2 (I think it has to be with OSX?), I don't know how pip packages are made but maybe that could have an effect even though I'm executing python3 commands?

erwincoumans, 2017-08-09:
It must be a problem very specific to your machine and multiple python installations. See also some hints here: https://stackoverflow.com/questions/34386527/symbol-not-found-pycodecinfo-getincrementaldecoder (for example the answer that starts with 'reinstall python'.

ahundt, 2017-08-09:
Yes it seems that way, always the worst type of problem.... The first thing I tried before the command line commands I posted was `brew reinstall python3` sorry I forgot to include that. Anyway, I'll check out the link and dig for a solution, or perhaps simply try running on another machine haha.

Thanks for the help!

erwincoumans, 2017-08-12:
It is much faster if you just install Anaconda, python 3.6. By default, Anaconda installs locally, not system-wide, so it is very convenient. It will add some path in ~/.back_profile, that you can undo.
After install Anaconda python 3.6, just run the following pip commands:
pip install pybullet
pip install tensorflow
pip install gym
pip install baselines

ahundt, 2017-08-13:
It ran now, thanks! 

It turned out some of my convenience tools were a bit too eager and I was messing up my modules/path somewhere, cleaning house on those fixed the problem but I don't have a very specific root cause, sorry about that. 

Regarding Anaconda, I'm actually using lots of custom packages including ROS and [Tasks (which has a newly released cython api that also integrates Eigen)](https://github.com/jrl-umi3218/Tasks). For that reason, switching to Anaconda seems like a larger project than my path problem once I narrowed it down.

A fun aside: While I was going through the python issues I also took the chance to set up [xonsh](http://xon.sh) shell since it lets me play with the pybullet environment interactively while still having access to normal bash like shell commands. :-)

closed 2017-08-13 by ahundt