bulletphysics/bullet3 #1243: Object Radius

romieval, 2017-07-28:
Bullet's code is giant. I cannot figure out a way to set a radius to btCollisionShape, at least not one things cannot go wrong. So please if that's a missing feature it would be great to have it.

So many thanks in advance!!!

erwincoumans, 2017-07-29:
You want to add some additional collision radius to existing shapes? There is only "setMargin", but this has various potential issues dependend on shape, local scaling etc. In addition, you need to manually update the rigid body (remove/reinsert to world, update inertia tensor etc). So there is no "easy way".

closed 2017-07-29 by erwincoumans

romieval, 2017-07-29:
Yes. That's exactly what I was looking for. Congrats for the code base. Thanks!