bulletphysics/bullet3 #1245: system malloc assertion

SamuelSJTU, 2017-07-30:
I try to run the helloworld code, but I got memory problem
`bullet_GL: malloc.c:2374: sysmalloc: Assertion `(old_top == (((mbinptr) (((char *) &((av)->bins[((1) - 1) * 2])) - __builtin_offsetof (struct malloc_chunk, fd)))) && old_size == 0) || ((unsigned long) (old_size) >= (unsigned long)((((__builtin_offsetof (struct malloc_chunk, fd_nextsize))+((2 *(sizeof(size_t))) - 1)) & ~((2 *(sizeof(size_t))) - 1))) && ((old_top)->size & 0x1) && ((unsigned long) old_end & pagemask) == 0)' failed.
`
this is because the problem of new and mallco

the code is  here

`    btAlignedObjectArray<btCollisionShape*> collisionShapes;

    ///create a few basic rigid bodies

    //the ground is a cube of side 100 at position y = -56.
    //the sphere will hit it at y = -6, with center at -5
    {
        btCollisionShape* groundShape = new btBoxShape(btVector3(btScalar(10.), btScalar(10.), btScalar(10.)));

        collisionShapes.push_back(groundShape);`

but I can run the example build in App_helloword
my cmakelist is here
`cmake_minimum_required(VERSION 2.4.3)
project(bullet_GL)

set(CMAKE_CXX_STANDARD 11)

set(SOURCE_FILES main.cpp)

INCLUDE_DIRECTORIES(
        "/usr/local/include/bullet"
)
LINK_DIRECTORIES("/usr/local/lib")
LINK_LIBRARIES(
        BulletDynamics
        BulletCollision
        LinearMath
)

add_executable(bullet_GL ${SOURCE_FILES})`
Can any have any idea to solve it

erwincoumans, 2017-07-30:
Never seen such warning, also there is no bullet_GL project, since the App_HelloWorld doesn't use OpenGL. Does the default App_HelloWorld CMake project trigger this warning/error?

SamuelSJTU, 2017-07-30:
My operating system is ubuntu 14.04 with g++ 4.8
bullet_GL is my own project and I copy the code in App_HelloWorld. I got no  error when I try to run App_HelloWorld. But after I install bullet and build my own project,  I got this error. And now I try the release version 2.8.6.1 with same operation， it seems alright.  Thank you for your reply.

pisuto, 2018-09-13:
sorry, now i meet the same problem although i run it in Ubuntu14.04.
can you explain the solution in detail?

cga-cmu, 2019-01-24:
I am getting a similar error:

cga@a1:~/bullet3-2.88/bullet/HelloWorld$ ./HelloWorld
collisionShapes.push_back(groundShape); fails
HelloWorld: malloc.c:2394: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
Aborted (core dumped)

Same situation. The app built by cmake in the distribution tree works fine, but when I copy that
directory out of the distribution tree and try to compile it as below, I get the error above.

g++ -I /usr/local/include/bullet -c HelloWorld.cpp
g++ -o HelloWorld HelloWorld.o -L /usr/local/lib/ -lBulletDynamics -lBulletCollision -lLinearMath 

ubuntu-16.04
gcc version 5.5.0 20171010 (Ubuntu 5.5.0-12ubuntu1~16.04)

erwincoumans, 2019-01-24:
Are you setting any flags to compile Bullet? A minimal repro case is needed. Compare your setup with regular working Bullet examples.

closed 2019-01-24 by erwincoumans

cga-cmu, 2019-01-24:
Issue is flag -DBT_USE_DOUBLE_PRECISION

This works
g++  -DBT_USE_DOUBLE_PRECISION -I/usr/local/include/bullet -c HelloWorld.cpp
g++   HelloWorld.o  -o HelloWorld -L /usr/local/lib/ -lBulletDynamics -lBulletCollision -lLinearMath 

This does not
g++ -I /usr/local/include/bullet -c HelloWorld.cpp
g++   HelloWorld.o  -o HelloWorld -L /usr/local/lib/ -lBulletDynamics -lBulletCollision -lLinearMath 

Minimum reproducible case (on my machine)
1) copy HelloWorld.cpp to somewhere.
2)  g++ -I /usr/local/include/bullet -c HelloWorld.cpp
g++ -o HelloWorld HelloWorld.o -L /usr/local/lib/ -lBulletDynamics -lBulletCollision -lLinearMath 
3)  ./HelloWorld
results in error message
HelloWorld: malloc.c:2394: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

erwincoumans, 2019-01-24:
That means that your system version of Bullet is compiled using -DBT_USE_DOUBLE_PRECISION.
How come so? And if so, you obviously have to use the same flags when using Bullet as you compiled Bullet. That's why it is easier to just embed the Bullet source files in your own app.