bulletphysics/bullet3 #1249: BulletIK

JulianYG, 2017-08-04:
**Problem 1: Parsing URDF**
It seems currently the Bullet IK can only parse arm urdf’s; that is, with the full robot model, it generates wrong IK solutions. The complete Sawyer robot model for example, has 20 joints, where 7 are activated joints and 13 are fixed joints. Bullet cannot construct a correct arm model from the full urdf, and the IK solutions are incorrect.

**Problem 2: Inverse Kinematics**
The robot with different joint states have different solutions for the same target pose. That is, the IK solutions are not state independent. 

It also needs multiple iterations to solve for one accurate solution. In bullet3/examples/pybullet/examples/inverse_kinematics.py, change use_real_time = 1 to 0, then the robot cannot reach to desired position at all.

The null space solution is very off. In bullet3/examples/pybullet/examples/inverse_kinematics.py, change use_null_space = 0 to 1, the actual and target lines are separated far away.

**Problem3: Rendering**
Depending on types of different hardware, the rendered color of the grippers are different. For example, a PR2 gripper image has red and green fingers, which is inconsistent with original yellow and green colors. This detriments the learning models based on visual inputs.

Pictures are attached in:
https://docs.google.com/document/d/1XH7GGaxPjahu-3z51t7aSw39DWgmjQ-NeFSvvyZYTdU/edit?usp=sharing

ahundt, 2017-09-11:
My project group has also noticed some problems with the IK. If you're willing to use external IK and only use bullet for the physics might I suggest https://github.com/jrl-umi3218/Tasks?

erwincoumans, 2017-09-12:
ahundt, JulianYG, please provide some URDF files that have issues with IK. That way we can resolve it, and people don't need to revert to an external IK solution.

Where can we find the Sawyer robot?

>>a PR2 gripper image has red and green fingers

The colors are not assigned in the URDF or SDF file, this can be easily fixed by the user.

Thanks for the feedback, this helps us improve pybullet etc.

JulianYG, 2017-09-12:
Yeah, I see some urdf files don't define colors in them, and they are related to the hardware that renders them. The sawyer urdf is here:
https://www.dropbox.com/s/aoxg0z0vheu02mp/sawyer_robot.zip?dl=0

Thanks to @ahundt for the link.

erwincoumans, 2017-09-12:
Thanks for providing the urdf, we'll look into it. There was a previous report of IK issues, but no URDF (https://github.com/bulletphysics/bullet3/issues/1199), was that your group Andrew?

ahundt, 2017-09-12:
Yes #1199 is my group. Here are the files zipped up [ur5_robotiq_85.zip](https://github.com/bulletphysics/bullet3/files/1297480/ur5_robotiq_85.zip).

arpit15, 2017-09-14:
Hey @erwincoumans
I am facing the same problem when loading baxter robot. The urdf and mesh files used in simulation are [testingforbaxterloading.zip](https://github.com/bulletphysics/bullet3/files/1304356/testingforbaxterloading.zip). Kindly let me know a solution to this issue if possible. 
Thank you
Arpit

erwincoumans, 2017-09-14:
Sounds good, thanks for sharing the problematic files. The person who will look into is is busy with some paper deadline at the moment. We will look at it soon.

erwincoumans, 2017-10-10:
Latest trunk should have working IK for those robots, tested with the Sawyer. Make sure to use latest pybullet 1.5.6 or newer, here is a test for Sawyer: goo.gl/m7S1GV
Also, the base needs to be fixed, either mass 0 or fixed to a link called "world" or rename "base" to "world".
Please check the UR5 and Baxter (provide some inverse_kinematics.py script modified for your robot. We cannot deal with XACRO, please convert to URDF.
Enjoy!

erwincoumans, 2017-10-10:
https://www.youtube.com/watch?v=JzEtoHvK11M
And an example script using the Sawyer: https://github.com/erwincoumans/pybullet_robots

closed 2017-10-10 by erwincoumans