bulletphysics/bullet3 #1251: how to use a sdf/world created in a gazebo in pybullet?

k-maheshkumar, 2017-08-11:
Hi,
Is it possible to use the world/sdf created from the gazebo in pybullet?. If so, can you please give me an example how to use it.

Thanks in advance.

erwincoumans, 2017-08-11:
You can use loadSDF to load a Gazebo created sdf file. Make sure to copy the file in the Bullet data folder, and use relative path names (no package:// etc) in the sdf file. See the two_cubes.sdf, this file was exported from Gazebo, with some minor manual changes.

erwincoumans, 2017-08-25:
Do you have a particular gazebo exported SDF file? If so, attach it and re-open the issue. I'll give some hints how to import it.

closed 2017-08-25 by erwincoumans

k123jack, 2017-10-04:
@erwincoumans 
Hi, I want to import a gazebo model, like coke_can. I copied it to Bullet data folder , changed path like model://coke_can/meshes/coke_can.dae in all sdf files to relative path like :  ../meshes/coke_can.dae .
But I still get :Cannot load SDF file.