bulletphysics/bullet3 #1253: contactManifold->getNumContacts() always 0 [assistance]

kenkit, 2017-08-14:
I am using bullet only for collision detection for now, I am self-taught programmer.
This is the only code that I could manage to get working. I can't find enough information.
I intend to use this code in a game I'm working on in Ogre3d.

Why dpes` getNumContacts()` always return zero even after collisions were detected ?

```
#include <iostream>
#include <fstream>
#include <vector>
#include <string>
#include <windows.h>
#include <btBulletDynamicsCommon.h>
#include "vld.h"

#define PRINTF_FLOAT "%7.3f"

constexpr float gravity = -10.0f;
constexpr float initialY = 10.0f;
constexpr float timeStep = 1.0f / 60.0f;
// TODO some combinations of coefficients smaller than 1.0
// make the ball go up higher / not lose height. Why?
constexpr float groundRestitution = 0.9f;
constexpr float sphereRestitution = 0.9f;
constexpr int maxNPoints = 500;

struct my_box {
	std::string box_name;
};

std::vector<btVector3> collisions;
void myTickCallback(btDynamicsWorld *dynamicsWorld, btScalar timeStep) {
	

    int numManifolds = dynamicsWorld->getDispatcher()->getNumManifolds();
    for (int i = 0; i < numManifolds; i++) {
		std::cout << "--------------------Collisions found--------------------" << std::endl;
        btPersistentManifold *contactManifold = dynamicsWorld->getDispatcher()->getManifoldByIndexInternal(i);
        // TODO those are unused. What can be done with them?
        // I think they are the same objects as those in the main loop
        // dynamicsWorld->getCollisionObjectArray() and we could compare
        // the pointers to see which object collided with which.
		
            const btCollisionObject *objA = contactManifold->getBody0();
            const btCollisionObject *objB = contactManifold->getBody1();

			my_box *kks = (my_box*)objA->getUserPointer();
			my_box *kkv = (my_box*)objB->getUserPointer();
			std::cout << "Collision detected BTWN :" <<kks->box_name<<" & "<< kkv->box_name;
		
			int numContacts = contactManifold->getNumContacts();
		std::cout << " CONTACTS:" << numContacts << std::endl;

		for (int j = 0; j<numContacts; j++)
		{
			btManifoldPoint& pt = contactManifold->getContactPoint(j);
			if (1/*pt.getDistance()<100.f*/)
			{
				const btVector3& ptA = pt.getPositionWorldOnA();
				const btVector3& ptB = pt.getPositionWorldOnB();
				const btVector3& normalOnB = pt.m_normalWorldOnB;
				my_box *kks = (my_box*)objA->getUserPointer();
				my_box *kkv = (my_box*)objB->getUserPointer();
				
			}
		}
    }
}

int main() {
    int i, j;

	my_box *mybox1 = new my_box;
	my_box *mybox2 = new my_box;
	mybox1->box_name = "Box1";
	mybox2->box_name = "Box2";

    btDefaultCollisionConfiguration *collisionConfiguration = new btDefaultCollisionConfiguration();
    btCollisionDispatcher *dispatcher = new btCollisionDispatcher(collisionConfiguration);
    btBroadphaseInterface *overlappingPairCache = new btDbvtBroadphase();
    btSequentialImpulseConstraintSolver* solver = new btSequentialImpulseConstraintSolver;
    btDiscreteDynamicsWorld *dynamicsWorld = new btDiscreteDynamicsWorld(dispatcher, overlappingPairCache, solver, collisionConfiguration);
	dynamicsWorld->setGravity(btVector3(0, gravity, 0));
    dynamicsWorld->setInternalTickCallback(myTickCallback);

	btVector3 fallInertia(0, 0, 0);
	btScalar mass = 1;
    // Ground.

		btDefaultMotionState* fallMotionState =
		new btDefaultMotionState(btTransform(btQuaternion(0, 0, 0, 1), btVector3(0, 0, 0)));
        btTransform groundTransform;
        groundTransform.setIdentity();
        groundTransform.setOrigin(btVector3(0, 0, 0));
        btCollisionShape* boxShape;
		boxShape = new btSphereShape(1);
		btRigidBody::btRigidBodyConstructionInfo fallRigidBodyCI(mass, fallMotionState, boxShape, fallInertia);
		
		

		btRigidBody *box1 = new btRigidBody(fallRigidBodyCI);
		box1->setUserPointer(mybox1);
		dynamicsWorld->addRigidBody(box1);
	
		btRigidBody  *box2 = new btRigidBody(fallRigidBodyCI);
		box2->setUserPointer(mybox2);
		dynamicsWorld->addRigidBody(box2);

    // Main loop.
   
	int y_box = 100;
	int y_box2 = 0;
	my_box *kks;
	btTransform k;
	for (int i = 0; i < 100;i++) {
		//std::cout << "bx 1:" << y_box<<" bx 2:"<<y_box2 << std::endl;
		const btCollisionObjectArray coA = dynamicsWorld->getCollisionObjectArray();
		//std::cout << "size:"<< coA.size() << std::endl;
		for (int j = 0; j < coA.size(); j++) {
			kks = (my_box*)coA.at(j)->getUserPointer();
			if (kks->box_name == "Box1") {
				
				k.setOrigin(btVector3((btScalar)0, (btScalar)y_box, (btScalar)0));
				coA.at(j)->setWorldTransform(k);
				std::cout << "SETTING BOX 1:" << coA.at(j)->getWorldTransform().getOrigin().getY() << std::endl;
			}
			if (kks->box_name == "Box2") {
				std::cout << "SETTING BOX 2:" << y_box2 << std::endl;
				k.setOrigin(btVector3((btScalar)0, (btScalar)y_box2, (btScalar)0));
				coA.at(j)->setWorldTransform(k);
			}
		}
		dynamicsWorld->stepSimulation(timeStep);
		y_box--;
		y_box2++;
    }

	Sleep(10000);
    // Cleanup.
    for (i = dynamicsWorld->getNumCollisionObjects() - 1; i >= 0; --i) {
        btCollisionObject* obj = dynamicsWorld->getCollisionObjectArray()[i];
        btRigidBody* body = btRigidBody::upcast(obj);
		my_box *kks = (my_box*)obj->getUserPointer();
        dynamicsWorld->removeCollisionObject(obj);
		delete body;
		delete kks;
    }
  
	delete boxShape;
    delete dynamicsWorld;
    delete solver;
    delete overlappingPairCache;
    delete dispatcher;
    delete collisionConfiguration;
	delete fallMotionState;
  
}
```

benelot, 2017-08-18:
Hello,

Can you please ask this question on the bulletphysics forum? The requests
here are usually only related to bugs or extensions. Thanks.

kenkit, 2017-08-19:
Okay, thanks I didn't know sorry.

closed 2017-08-19 by kenkit