bulletphysics/bullet3 #1255: Cannot Connect to Physics Server

kerenstaro, 2017-08-15:
Hello,
We are currently following the instructions from [here.](https://github.com/bulletphysics/bullet3/blob/master/README.md) on how to build the Windows Virtual Reality sandbox for HTC Vive and Oculus Rift.  So far we have downloaded the master version, and are now trying to run the commands:

`import pybullet as p`
`p.connect(p.SHARED_MEMORY)`

However when we do so, it stalls (does not terminate). We also tried the two other options using UDP and TCP, but they did not work for us (failed connection).

Would appreciate any help in how to debug or go about this problem.
Thanks!

erwincoumans, 2017-08-15:
Did you run the VR sandbox first? Are you using the latest github version of Bullet? How did you install pybullet? Are you using the 64bit bit version of pybullet and 64bit version of the VR server? The bit-ness has to be the same (both pybullet/Python and VR server 32bit or both 64bit, not a mix)

kerenstaro, 2017-08-16:
Hi @erwincoumans ,
Thanks so much for drawing this to our attention. Now everything uses a 64bit version and is working!

closed 2017-08-16 by kbtruth