bulletphysics/bullet3 #1262: Scaling ghost objects's shapes doesn't seem to work

ongamex, 2017-08-19:
If a call shape->setLocalScaling(...) on a shape that is attached to a btGhostObject the scaling won't take effect.

This works correctly with rigid bodies.

I'm on commit tagged as "2.86.1"

ongamex, 2017-08-26:
For those who have the same issue. I cound't fix the btGhostObject shape scaling. 
Instead now I'm using a standard rigid body with btCollisionObject::CF_NO_CONTACT_RESPONSE. This flag will still let bullet compute the contact manifolds, but they wont be used to resolve any collisions.

And on each step I loop trough all contact manifolds and find the ones that have my rigid body with that flag in it.

ongamex, 2017-08-26:
Otherwise, tell me if need more info about the issue,

benelot, 2017-08-26:
Can you move this issue to the bulletphysics forum? This is meant to be the
place where you post a problem if you are pretty sure that it is an actual
bug.

ongamex, 2017-08-30:
Well, pretty sure that it is a bug

closed 2021-04-26 by erwincoumans