bulletphysics/bullet3 #1266: RigidBody scaling

AndreaCatania, 2017-08-22:
I'm implementing a wrapper of Bullet Physics engine to Godot game engine, here the PR https://github.com/godotengine/godot/pull/10013.
It's almost done, I just need fix some problems.
One of these is that I need set the scaling of a RigidBody.
I know that I can use setLocalScaling on the shape but I don't want use it since the shape may be reused with different scale by another RigidBody.

Each RigidBody has btCompoundShape where I can set a shape with its transform but I've noticed that this transform not consider the scale.

So the question is that how can I scale the btRigidBody?

I think is possible since this is a common feature.

ongamex, 2017-08-22:
Can you just add the scaling to the equation when reusing?

AndreaCatania, 2017-08-22:
What do you mean specifically? can you give me an example?

ongamex, 2017-08-22:
Well you've said that you're reusing the collision shape. And when you do a lookup if a shape is already registered, you check the local scaling of the shape, if the scaling is set to your desired value, if not create a new shape with the scaling you need.

erwincoumans, 2017-08-22:
For most of the basic shapes  (btSphereShape, btBoxShape etc) and btCompoundShape it is likely easier to disable shape sharing. For btConvexHullShape and btBvhTriangleMeshShape there are ways to share the vertex data (biggest data consumer) and don't share the actual shape. For example use btConvexPointCloudShape (allows vertex sharing) and btScaledBvhTriangleMeshShape (other options are possible too).

AndreaCatania, 2017-08-28:
@erwincoumans @ongamex Thanks for this explanation, I've implemented it.
For a case I've found that if I set the scale to the transform in the compound shape the shape is also scaled.

This behavior is not intented to work or I can use it? If I can use it is shearing shape shearing allowed?

erwincoumans, 2017-08-28:
No shearing is supported, only rigid body transforms (position and orientation quaternion).

I wouldn't add scaling in the transform: just use btCollisionShape local scaling and possibly adjust the transforms of the compound shape manually if needed.

AndreaCatania, 2017-08-28:
@erwincoumans Ok Thanks you!

closed 2017-08-28 by AndreaCatania

tagcup, 2017-08-29:
@erwincoumans Godot has a tree-like node structure, so the global transform is in general a product of multiple local transformations in general. While each local basis is just a rotation and scale, Bi = Ri.Si, a composite basis can have things like shear (example: a child node rotated around z-axis by 45 degrees and a parent node with scaling 2,1,1 results in a general affine matrix which can't be decomposed into a global rotation and a scaling).

Is there any way to represent such objects in Bullet?