bulletphysics/bullet3 #1275: pip install pybullet on mac/linux can't run examples, cant find  URDF files

jtoy, 2017-08-26:
I tried on ubuntu 14.04 and mac 10.5.5. I run "pip install pytorch"  I am using anaconda  with python 3 on both OSes.  I try to run any of the examples on both OSes and they all die  on the line that calls loadURDF wit hthe error:
pybullet.error: Cannot load URDF file.
I see that both OSes install urdf files in the data dir like so:
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/block.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/cartpole.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/plane.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/sphere2.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/sphere2red.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/kuka_iiwa/model.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/kuka_iiwa/model_for_sdf.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/kuka_iiwa/model_free_base.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/kuka_iiwa/model_vr_limits.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/quadruped/minitaur.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/quadruped/minitaur_fixed_all.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/quadruped/minitaur_fixed_knees.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/quadruped/quadruped.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/racecar/racecar.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/racecar/racecar_differential.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/table/table.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/tray/tray.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/tray/tray_textured2.urdf
/home/jtoy/anaconda3/lib/python3.6/site-packages/pybullet_envs/data/tray/traybox.urdf

erwincoumans, 2017-08-26:
Could it be a permissions issue? sudo chown -R $USER /home/jtoy/anaconda3

What command-line are you exactly running to run the examples?

erwincoumans, 2017-08-26:
pip install pytorch? Did you mean pip install pybullet? Without the sudo?

erwincoumans, 2017-08-26:
Also, I just uploaded version pybullet-1.2.5, could you try reinstall as well?

erwincoumans, 2017-08-26:
As a test, after pip install pybullet (version 1.2.5 or later, pip uninstall pybullet first) use this:

python3
import pybullet_envs.examples.enjoy_TF_HumanoidBulletEnv_v0_2017may
pybullet_envs.examples.enjoy_TF_HumanoidBulletEnv_v0_2017may.demo_run()

Let me know if that works for you (or not)

jtoy, 2017-08-26:
* yes, I meant to say "pip install pybullet".   
* its  not an issue with permissions as I install with the local user and anaconda and no sudo.
* i just copy/paste the source  of from here https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/examples  into my local directory and run "python SCRIPT.py" and the window appears for a second and dies with the urdf error.  I also tried copying those urdf files to my local directory and also into local_directory/data and get the same issue.  My guess is I shouldn't copy them like that.
* I installed pybullet 1.2.5 and get the same issue on both mac and linux

* this command works "python3
import pybullet_envs.examples.enjoy_TF_HumanoidBulletEnv_v0_2017may
pybullet_envs.examples.enjoy_TF_HumanoidBulletEnv_v0_2017may.demo_run()" Looks cool :)
* this command works as well:  pip show -f pybullet

How can I know where pybullet's data directory is, is there a way to manually set it?

erwincoumans, 2017-08-26:
If enjoy_TF_HumanoidBulletEnv_v0_2017may works, it is all fine. I haven't ported the KUKA assets yet.

What example fails exactly, and copy/paste the full output please.

jtoy, 2017-08-26:
I got it all working, I have to run the examples from inside the git cloned repo. 
I think it might be good to make the examples be able to run standalone, or maybe add a README in the pybullet examples dir.  The problem is you cant really run "pip install pybullet" and then run any of the examples that require urdf files unless you clone the whole repo.
 I read through the pybullet google doc, such fun potential to use for machine learning.
 is the goal for the pybullet lib to get to feature parity with C API? Will there be official python docs?
Looks like the python stuff is still work in progress. Would love to help out, taking off on vacation tomorrow, back in a week!

erwincoumans, 2017-08-26:
>>The problem is you cant really run "pip install pybullet" 
>>and then run any of the examples that require urdf files unless you clone the whole repo.

Yes, you did run some example after running pip install pybullet. You don't need the github repo at all to run the GYM environments. You confirmed that this example worked for you:

import pybullet_envs.examples.enjoy_TF_HumanoidBulletEnv_v0_2017may
pybullet_envs.examples.enjoy_TF_HumanoidBulletEnv_v0_2017may.demo_run()"

One more try: can you please report the command that you tried to run (without the github repo), and provide the console/error log?

The pybullet quickstart guide IS the official python docs. Can you explain what makes it not official or incomplete?

The pybullet API is pretty mature at the moment, and various teams in Google use it for machine learning research.

The pybullet_envs are the GYM environments, those are brand new and not documented yet and not all examples/assets are converted to be run within the pip installed version.

Thanks for trying it and for reporting back.

erwincoumans, 2017-08-26:
I uploaded pybullet version 1.2.6. Please have a good look at this commit info and please do NOT need to use the github repository, just pip install pybullet.

pip install pybullet

Enjoy pretrained environments:

python -m pybullet_envs.examples.enjoy_TF_AntBulletEnv_v0_2017may
python -m pybullet_envs.examples.enjoy_TF_HalfCheetahBulletEnv_v0_2017may
python -m pybullet_envs.examples.enjoy_TF_AntBulletEnv_v0_2017may
python -m pybullet_envs.examples.enjoy_TF_HopperBulletEnv_v0_2017may
python -m pybullet_envs.examples.enjoy_TF_HumanoidBulletEnv_v0_2017may
python -m pybullet_envs.examples.enjoy_TF_InvertedDoublePendulumBulletEnv_v0_2017may
python -m pybullet_envs.examples.enjoy_TF_InvertedPendulumBulletEnv_v0_2017may
python -m pybullet_envs.examples.enjoy_TF_InvertedPendulumSwingupBulletEnv_v0_2017may
python -m pybullet_envs.examples.enjoy_TF_Walker2DBulletEnv_v0_2017may

Run some gym environment test:
python -m pybullet_envs.examples.racecarGymEnvTest

Using OpenAI baselines DQN training:

train:
python -m pybullet_envs.examples.train_pybullet_cartpole
python -m pybullet_envs.examples.train_pybullet_racecar

(the training will save a .pkl file with the weights, leave it running for a while, it terminates when it reaches some reasonable reward)

enjoy:
python -m pybullet_envs.examples.enjoy_pybullet_cartpole
python -m pybullet_envs.examples.enjoy_pybullet_racecar

For your own learning/training, create/Import a specific Gym environment:

python
import pybullet_envs
env = gym.make("AntBulletEnv-v0")
env = gym.make("HalfCheetahBulletEnv-v0")
env = gym.make("HopperBulletEnv-v0")
env = gym.make("HumanoidBulletEnv-v0")
env = gym.make("Walker2DBulletEnv-v0")
env = gym.make("InvertedDoublePendulumBulletEnv-v0")
env = gym.make("InvertedPendulumBulletEnv-v0")
env = gym.make("MinitaurBulletEnv-v0")
env = gym.make("RacecarBulletEnv-v0")
env = gym.make("KukaBulletEnv-v0")
env = gym.make("CartPoleBulletEnv-v0")

If you want to enable human/GUI rendering in a Gym-created environment, call the env.render(mode="human") BEFORE the first env.reset.

jtoy, 2017-08-27:
the pretrained environments work from your comment above! What doesnt work is using the examples from source:

 wget https://github.com/bulletphysics/bullet3/raw/master/examples/pybullet/examples/hello_pybullet.py
jtoy@MacBook-Pro-6:~/sandbox/bullet_test $ python hello_pybullet.py
pybullet build time: Aug 27 2017 06:05:02
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce GT 750M OpenGL Engine
Version: 4.1 NVIDIA-10.4.2 310.41.35f01
GLSL: 4.10
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0
MotionThreadFunc thread started
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,139]:

b3Printf: URDF file 'plane.urdf' not found

Traceback (most recent call last):
  File "hello_pybullet.py", line 7, in <module>
    planeId = p.loadURDF("plane.urdf")
pybullet.error: Cannot load URDF file.
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

wget https://raw.githubusercontent.com/bulletphysics/bullet3/master/examples/pybullet/examples/test.py

python test.py
pybullet build time: Aug 27 2017 06:05:02
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce GT 750M OpenGL Engine
Version: 4.1 NVIDIA-10.4.2 310.41.35f01
GLSL: 4.10
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0
MotionThreadFunc thread started
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,139]:

b3Printf: URDF file 'plane.urdf' not found

Traceback (most recent call last):
  File "test.py", line 6, in <module>
    pybullet.loadURDF("plane.urdf",0,0,-1)
pybullet.error: Cannot load URDF file.
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

erwincoumans, 2017-08-27:
Ah, pybullet_env only exposes gym environments. It has a small subset of private data (urdf, obj) ONLY for those gym envs. If you want to run the regular github examples, just checkout the full tree. I'll clarify this in the future pybullet Gym chapter of the quickstart guide.

jtoy, 2017-08-27:
Cool, yeah, that was what I meant but adding that info in the docs/README. Thanks!

-- Sent from my mobile

erwincoumans, 2017-08-28:
OK, pybullet 1.2.7 addresses the issue. There is the pybullet_data package, that has the getDataPath() method. I didn't convert all examples to use it, just 'test.py' and 'racecar.py'.
There is also a new pybullet.setAdditionalSearchPath.

jtoy, 2017-09-04:
Just got back. I did some tests on the examples with pybullet 1.2.9 and they all seem to work with the new changes now, thank you.

closed 2017-09-04 by jtoy