bulletphysics/bullet3 #1276: URDF loading problems when trying to port the Atlas robot gym env from robot school

benelot, 2017-08-26:
I tried to port it to pybullet, however, I can notload the atlas robot from urdf. I removed the package definitions and restructured it so that it would link into each other properly, but I think it has problems loading the contained dae files. Is this known to be a problem? Is there are a measure to fix this or is the already a model somewhere that works?

erwincoumans, 2017-08-26:
The DAE loader is very primitive, likely doesn't accept the file. Best would be to convert the assets to Wavefront .OBJ.

benelot, 2017-08-26:
Thanks for the info. I will try that.

erwincoumans, 2017-08-26:
By the way, the reorganization of the Gym environments is close to done

1) pybullet connection/disconnection needs to happen in the environment

2) if you want to enable GUI/human rendering, call the env.render(mode="human") BEFORE the first env.reset

https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/examples/enjoy_TF_AntBulletEnv_v0_2017may.py

You can try it out by pip install pybullet. Then you can run the examples, without requiring to have the Bullet repository installed.

This commit shows some examples how to use the Gym environments after pip install pybullet:

https://github.com/bulletphysics/bullet3/pull/1279/commits/51b7e1040f0581dbb5a2e3f2eeb38ac55f112b18

benelot, 2017-08-26:
I realized the 'close to done' after I stated merging your reorganization with my new stuff today and then saw that you were working on some other restructuring changes. However, it seems that the last merge should not be a large thing anymore. Thanks for the update, I will make my contributions comply with the above described structure.

erwincoumans, 2017-08-26:
Thanks, and please try it out. If things don't work, let me know. Try it using pure "pip install pybullet" (version 1.2.6 or later) outside/without regular Bullet github and inside. See that commit for more into how to try it out (https://github.com/bulletphysics/bullet3/commit/51b7e1040f0581dbb5a2e3f2eeb38ac55f112b18)

erwincoumans, 2017-09-05:
Roboschool is uses Bullet to load the Atlas URDF file using the loadURDF command. Are you using a different URDF file?

I just modified Roboschool to connect to shared memory to the latest version of Bullet ExampleBrowser, and dumped the state using the pybullet.saveWorld command, here is the output with the actual file names:

import pybullet as p
cin = p.connect(p.SHARED_MEMORY)
if (cin < 0):
    cin = p.connect(p.GUI)
objects = p.loadMJCF("/Users/erwincoumans/dev/roboschool/roboschool/mujoco_assets/ground_plane.xml")
ob = objects[0]
p.resetBasePositionAndOrientation(ob,[0.000000,0.000000,0.000000],[0.000000,0.000000,0.000000,1.000000])
objects = [p.loadURDF("/Users/erwincoumans/dev/roboschool/roboschool/models_robot/atlas_description/urdf/atlas_v4_with_multisense.urdf", 0.052097,-0.109907,0.853118,0.020115,-0.042640,-0.126953,0.990788)]
ob = objects[0]
jointPositions=[ 0.664932, 0.257907, -0.020424, -1.082520, -1.379381, 1.127177, 1.174691, 0.439071, -0.607822, 0.001000, 0.396434, -0.685561, 1.496594, -0.000015, -2.220817, 0.838156, -0.827978, 0.001015, 0.183998, -0.031814, -0.119002, 0.816444, -0.610341, -0.003797, 0.008446, -0.061425, 0.183882, 0.564570, -0.584537, -0.009632 ]
for jointIndex in range (p.getNumJoints(ob)):
        p.resetJointState(ob,jointIndex,jointPositions[jointIndex])

p.setGravity(0.000000,0.000000,-9.800000)
p.stepSimulation()
p.disconnect()

benelot, 2017-09-05:
It works out of the box, it was a file path issue.

closed 2017-09-05 by benelot