bulletphysics/bullet3 #1277: Linux: ExampleBrowser uses 1 single core at 100%

benelot, 2017-08-26:
Even if I nothing is loaded and nothing is connected to the example browser, it is pretty processor intensive. Is there a reason for it? Is/was there some work on that (I am not sure if I have the latest version)? It seems to run at full power all the time...

Paul17041993, 2017-08-26:
Yes, it runs at max speed, doesn't use vsync at all, so what you're seeing is expected. I don't think it'd really be a good idea to change this in the demo explorer, but a checkbox on the window to enable vsync might be useful...

Sometimes vsync or fps limits can be a bad thing, it can reflect poorer than usual performance and in some cases may not work properly at all (eg; mesa drivers).

benelot, 2017-08-26:
Well ok, that sounds like a good reasoning for the current situation.
Thanks for the headsup!

closed 2017-08-28 by benelot