bulletphysics/bullet3 #1281: Object moving on its own, without any input (only gravity)

jplobos, 2017-08-28:
Hello,

I have attached a cylinder mesh to the end effector of a robot arm via the 'createMagnetConstraint' API. When running the application, the cylinder and by extension the end effector move by themselves without any input or forces (only gravity is 'on'). The cylinder, attached to the end effector, moves 'up' in the positive Z direction rolling about the magnet connection.

Any idea what might be happening? Also apologies if this isn't the correct venue for this question. 

Thanks!

erwincoumans, 2017-08-29:
I have never heard of a createMagnetConstraint API. Can you provide some easy to reproduce C++ and/or pybullet reproduction case?

jplobos, 2017-08-29:
Apologies, it's the createConstraint API on pg. 24 of the pybullet quickstart guide.  

'https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/

Unfortunately, I'm unable to provide the data files (URDFs) or much code, other than the direct call:

c = pybullet.createConstraint(id_robot, 6, id_cylinder, pybullet.JOINT_FIXED, [0, 0, 0], [0, 0, 0], [0, 0, 0], [0, 0, 0], [0, 0, 0])
pybullet.changeConstraint(c, maxForce=1000)

I suspect it might be something to do with the coordinate frames passed in, but I'm not sure as I'm very new to pybullet. 

Thanks for your help!

jplobos, 2017-08-29:
This line

c = pybullet.createConstraint(id_robot, 6, id_cylinder, pybullet.JOINT_FIXED, [0, 0, 0], [0, 0, 0], [0, 0, 0], [0, 0, 0], [0, 0, 0])

should read:

c = pybullet.createConstraint(id_robot, 6, id_cylinder, 0,  pybullet.JOINT_FIXED, [0, 0, 0], [0, 0, 0], [0, 0, 0], [0, 0, 0], [0, 0, 0])

closed 2021-04-26 by erwincoumans