bulletphysics/bullet3 #1293: loading stl files from pybullet possible?

jtoy, 2017-09-04:
Is it possible to directly load stl files in pybullet?
I am trying to take a random stl file like this one:   https://cl.ly/382g2j2s090M (from thingiverse ) and load it into a pybullet environment. 
I saw in the  C code there is a function called LoadMeshFromSTL, but that doesnt seem to be usable from pybullet.   It seems loadURDF can't load stl files as well.

erwincoumans, 2017-09-04:
You can create a simple URDF file with a visual or collision mesh that points to a .STL file.

Alternatively, you can use the pybullet.createCollisionShape with GEOM_MESH type pointing to that STL file (or .OBJ). Then use the pybullet.createMultiBody to create an actual object.

jtoy, 2017-09-07:
I got this working  using the first method  you suggested. I know stl is not a core part of bullet, but is it worth it to put some docs/notes for other people who run into a similar issue?

erwincoumans, 2017-10-11:
There is a new API pybullet.createVisualShape. Right now, it only load Wavefront OBJ, but we can add STL support later (binary STL)

ugurkanates, 2020-04-20:
I wish this issue didnt went silent

csingh27, 2020-05-10:
@erwincoumans Hi I am unable to import OBJ files using createVisualShape. I have tried with different files.
I use it like the following : 
visualShapeId = p.createVisualShape(shapeType=p.GEOM_MESH, fileName="pawn.obj")
Is there something I am doing wrong with the usage?
Thanks

lorepieri8, 2021-01-21:
Any plan to close this issue? It would be pretty handy.

luator, 2021-03-03:
I think this is already resolved since quite a while.  At least I'm using STL meshes with pyBullet and everything is working.

lorepieri8, 2021-03-03:
How are you loading them? Right now I'm calling them from a URDF file. 
Ideally there could be an API like: p.loadSTL(...)

luator, 2021-03-03:
For an object with collisions:
```python
shape_id = pybullet.createCollisionShape(
    shapeType=pybullet.GEOM_MESH,
    fileName="path/to/mesh.stl",
)
body_id = pybullet.createMultiBody(
    baseCollisionShapeIndex=shape_id,
    basePosition=(0, 0, 0),
    baseOrientation=(0, 0, 0, 1),
)
```
For an object without collisions (i.e. only for visualisation):
```python
shape_id = pybullet.createVisualShape(
    shapeType=pybullet.GEOM_MESH,
    fileName="path/to/mesh.stl",
)
body_id = pybullet.createMultiBody(
    baseVisualShapeIndex=shape_id,
    basePosition=(0, 0, 0),
    baseOrientation=(0, 0, 0, 1),
)
```

erwincoumans, 2021-03-03:
Indeed, STL files can be loaded using the createCollisionShape and createVisualShape API, and also be referenced in a URDF file.
[teststl.zip](https://github.com/bulletphysics/bullet3/files/6077867/teststl.zip)

Wavefront OBJ files are better: 
1) they can have textures (for visualShape)
2) they can store a convex decomposition (for collision shapes), so you can have moving concave colliders.
See attached teststl.zip, the testobj.py script shows a textured obj and also a convex decomposition (use 'w' to see the wireframe, and compare teststl with testobj. 
You can use Blender to create a manual convex decomposition (go to edit more, select parts (faces, vertices, edges) and press 'p' to make the selection into a separate obkject in the obj file. When export to obj, you can choose to have objects being separate. PyBullet will properly store them in a compound shape (btCompoundShape) under the hood.

closed 2021-03-03 by erwincoumans