bulletphysics/bullet3 #1295: pybullet.changeVisualShape issue with rgba

erwincoumans, 2017-09-05:
This method causes unpredictable segfaults when setting colors of objects. For example, try load plane_with_restitution, set a random rgba color for link -1. Then load cube_small, set a random color for link -1. This sometimes causes segfault.

erwincoumans, 2017-09-05:
Attached script works fine. Can you try attached script? If it works fine, please modify until it doesn't work for you.

[setVisShape.py.zip](https://github.com/bulletphysics/bullet3/files/1277773/setVisShape.py.zip)

JulianYG, 2017-09-05:
I cannot replicate the problem individually in a separate script, but I have located my problem to be the line with setting cube_small color at link -1. It segfaults 6 out of 10 times, very unpredictable. However, I have another problem with changing texture as well. The changeTexture.py cannot function properly, both planes appear white and cannot render the tex256.png. The snippet is attached:

import pybullet as p 

p.connect(p.GUI)
p.setRealTimeSimulation(1)

c = p.loadURDF('cube.urdf', (0, 0, 1), useFixedBase=True)

x = p.loadTexture('tex256.png')

p.changeVisualShape(c, -1, x)
p.getCameraImage(300,300,renderer=p.ER_TINY_RENDERER)
while 1:
	p.stepSimulation()

JulianYG, 2017-09-06:
I'm using mac 10.12.6 Sierra, with python 2.7 64bits. I'm still looking for a way to regenerate the problem, but my code was in a huge system and it's hard to track what exactly happened. I basically have a body class that represents the cube, and I use some setters/getters to set/query the state of the cube, which causes the error.

I also noticed that sometimes if wrong argument types are given to bullet functions, python will bark errors at some other random functions about "TypeError: required a float". I will try to look into those and write some scripts.

Sorry, it was not the cube, the texture on the plane does not work:

import pybullet as p 

p.connect(p.GUI)
p.setRealTimeSimulation(1)

f = p.loadURDF('../bullet3/data/plane_with_restitution.urdf', useFixedBase=True)
x = p.loadTexture('tex256.png')

p.changeVisualShape(f, -1, x)
p.getCameraImage(300,300,renderer=p.ER_TINY_RENDERER)
while 1:
	p.stepSimulation()

[image]

JulianYG, 2017-09-06:
I just found a case with the problem. It's a simple modification on the pybullet/examples/vr_kuka_setup.py. The code is attached here:

https://www.dropbox.com/s/2489h34ponpzeqj/test.py?dl=0

erwincoumans, 2017-09-07:
[test.py.zip](https://github.com/bulletphysics/bullet3/files/1285581/test.py.zip)

Great, thanks for spending the time to recreate the issue, I appreciate it and have a look at it soon!

erwincoumans, 2017-09-07:
Does it also fail if you point to the texture using the proper relative path (or absolute path)?
loadTexture('../bullet3/data/tex256.png')

JulianYG, 2017-09-07:
x = p.loadTexture('/Users/bullet3/data/tex256.png')

p.changeVisualShape(cube[0], 0, textureUniqueId=x)
# p.changeVisualShape(cube[0], -1, textureUniqueId=x)
p.changeVisualShape(plane[0], -1, textureUniqueId=x)

I changed it to the absolute path, and now the floor loads successfully, except the cube.

erwincoumans, 2017-10-11:
I made some fixes recently and ran your script/repro case and see no crash.
Can you confirm the issue is resolved? Thanks!

JulianYG, 2017-10-11:
I verified and the problem is solved. Thanks!

erwincoumans, 2017-10-11:
Great, thanks for the report, repro and verification! Let's close it then.

closed 2017-10-11 by erwincoumans