bulletphysics/bullet3 #1298: Bullet do not set FIXED_CONSTRAINT_TYPE to fixed constraints

1vanK, 2017-09-07:


erwincoumans, 2017-09-07:
Please provide a reproduction case.

1vanK, 2017-09-07:
src\BulletDynamics\ConstraintSolver\btFixedConstraint.cpp
```
btFixedConstraint::btFixedConstraint(btRigidBody& rbA,btRigidBody& rbB, const btTransform& frameInA,const btTransform& frameInB)
:btGeneric6DofSpring2Constraint(rbA,rbB,frameInA,frameInB)
{
	setAngularLowerLimit(btVector3(0,0,0));
	setAngularUpperLimit(btVector3(0,0,0));
	setLinearLowerLimit(btVector3(0,0,0));
	setLinearUpperLimit(btVector3(0,0,0));
}
```

just set to D6_SPRING_2_CONSTRAINT_TYPE

erwincoumans, 2017-09-07:
Do you have a suggested fix/patch?

1vanK, 2017-09-07:
I wanted to add fixed constraint to Urho3D in the image and likeness of other constraints. They use type in this way
https://github.com/urho3d/Urho3D/blob/master/Source/Urho3D/Physics/Constraint.cpp#L427

But personally, I only recently began to get acquainted with Bullet library so I can not provide any patch

erwincoumans, 2017-09-07:
You could just use the btGeneric6DofSpring2Constraint directly, and set the linear/angular limits to zero, that would workaround if there is any issue.

Cool, thanks for the report!

erwincoumans, 2017-09-07:
I moved to pybullet and a C-API, that allows also multibody in an easier way (btMultiBody) instead of btRigidBody and btTypedConstraints. See also the pybullet quickstart guide at http://pybullet.org if you are interested. The underlying C-API is abstracted and very different from the Bullet C++ API.

closed 2021-04-26 by erwincoumans