bulletphysics/bullet3 #1300: usability issues

jtoy, 2017-09-07:
* there is no list of keyboard/mouse commands for controlling the simulation. maybe put in a documentation section for this. 

* when the simulation is loaded, you dont know if your object got loaded. maybe show a list of objects and you can click on an item in the list to highlight it

* a couple of fully documented examples might be useful for novices

* only binary stls load

* maybe make docs runnable  where you can click a button and see it work.  

here are some examples of popular documentation:  https://stripe.com/docs/

ahundt, 2017-09-11:
have you seen http://pybullet.org?

jtoy, 2017-09-11:
yes I have, some of the usability issues I outlined are for the GUI when running a simulation.

jtoy, 2017-09-12:
* in the docs,there are several load methods, loadURDF, loadBullet, loadSDF,loadMJCF, what  is the preferred/best method?  docs should mention this. docs should also mention the tradeoffs of the different methods.

erwincoumans, 2017-09-12:
There is no preferred/best really, they are existing file formats, from ROS/Gazebo (URDF, SDF) and MuJoCo (MJCF). loadURDF is probably best supported, but it is only a single robot. loadSDF allows 
multiple robots. URDF and SDF are flat files, while MJCF is hierarchical.

Also we only supported a small, undocumented subset of URDF, SDF and MJCF.

Just like loading .gif, .png for images, it just depends.

BluePrintRandom, 2017-09-12:
I wonder if gIFT should be able to handle some physics data?

erwincoumans, 2017-09-12:
We develop pybullet for our needs in robotics and machine learning, and the primary file formats are URDF, SDF and MJCF.  We don't develop formats convenient for other people, and prefer not a wild-grow of file formats.

BluePrintRandom, 2017-09-12:
yeah, It's just blender, ue4 etc will support gIFT, and panda3D, godot etc

but like I said, I don't even know if it can hold the data.

(just seemed to be everyone pushing that way at once)

Thank you again Erwin for all the amazing work you do!

jtoy, 2017-09-12:
*  I see there is a function getMouseEvents, but no example, it would be great to add a mouse example to show people how to use it.

jtoy, 2017-09-12:
On this item I added : "when the simulation is loaded, you dont know if your object got loaded. maybe show a list of objects and you can click on an item in the list to highlight it"   I think it would also be extremely useful if when in debug mode, if you mouse over/touch an object or link, the item in the list highlights.

jtoy, 2017-09-16:
* better error messages, calling some functions with not the right amount of arguments returns an error of : "pybullet.error: Not connected to physics server." instead of something more accurate like "wrong arguments"    see https://github.com/bulletphysics/bullet3/issues/1319 for a more detailed example

benelot, 2017-09-17:
Are you interested in making some examples for the ExampleBrowser? I made
some in the paat for the bullet 2 API, so you could build up some for the
python interface.

jtoy, 2017-10-24:
* Ability to change from the code, the title of the window so it doesn't always say "Bullet Physics ExampleBrowser using OpenGL3+. Release Build"
 https://cl.ly/1E1w0B1r1Z3d

erwincoumans, 2020-04-18:
Closing this old issue, please use the PyBullet Quickstart Guide and the [examples here](https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/examples).
The getMouseEvents has an example linked in those docs.

After objects are loaded, you can use getNumBodies and getBodyInfo, getJointInfo to see what was loaded.

The title is hardcoded, some visibility for PyBullet in the GUI is desired.

Only binary STL files indeed. Recommended format is URDF with primitives (box, sphere etc) or Wavefront obj files with mtl/png files.

~~~~~~~~~~~~~
pip install pybullet --upgrade --user
~~~~~~~~~~~~~

closed 2020-04-18 by erwincoumans

liusida, 2020-08-15:
A quick update for `xacro`.

Now the `xacro` has 2.0 version, so the `xacro_standalone.py` mentioned doesn't work anymore. However, we could download the new version of `xacro` here:
https://github.com/ros-gbp/xacro-release/releases

and use a `pip install`, and we can use this Python package to produce `urdf` file.

So we can easily write a small Python tool (basically just modify the `main()` function in xacro package code), converting `xacro` to `urdf` first and calling `p.loadURDF` to get things done.

> Also note that URDF files online are often obfuscated in the .xacro format, so you need to install ROS (take a vacation for that) or use a rudimentary [xacro_converter.py](https://github.com/bulletphysics/bullet3/blob/master/data/xacro_standalone.py), or do it by hand.
> Best bet is just to be patient, become familiar with one (or more) of the file formats and deal with it. Understanding the underlying data format is useful in the long run. Welcome to robotics land!
>

Pang-Yatian, 2020-09-16:
> A quick update for `xacro`.
>
> Now the `xacro` has 2.0 version, so the `xacro_standalone.py` mentioned doesn't work anymore. However, we could download the new version of `xacro` here: [...]

Hi liusida,
I am not quite familiar with xacro, and what you said does not make sense to me. May I ask is it convenient to provide a simple example? Thank you in advance.