bulletphysics/bullet3 #1305: resetting simulation state is extremely slow, consider snapshotting without shutting down?

ahundt, 2017-09-11:
All the examples seem to completely restart the pybullet simulations and re-load from disk, such as during a call to `resetSimulation()` which unfortunately can consume the bulk of the time for each training run when executing simple tasks. 

Could you consider adding a `stateid = getSimulationSnapshot()` function that will store state and a `setSimulationSnapshot(stateid)` which will restore state? 

This could assume the objects present in the world do not change, they just return to their snapshotted configuration/position/orientation/velocity/joint angles etc. This could substantially accelerate the training process. Thanks for your consideration!

erwincoumans, 2017-09-12:
Yes, I will add this in the future, but it is going to take a while. You can keep this topic open.

erwincoumans, 2017-09-12:
Are you using DIRECT connection mode or GUI? Loading models in GUI mode and VR is slower, and can be sped up by temporary disabling graphics. Also, can you provide some performance profile, by starting the log before loading, and stopping the log after loading is done and attach the json file?

logId = p.startStateLogging(p.STATE_LOGGING_PROFILE_TIMINGS,"loadTimings.json")
#loadURDF, loadMJCF, loadSDF etc
p.stopStateLogging(logId)

cpaxton, 2017-09-12:
Just to add to this since @ahundt is discussing my simulation code here -- the reason this is an issue is because I saw simulation instabilities build up if I manually reset the positions of all entities in the world.

Unfortunately it's pretty hard for me to build an isolated case for this.

cpaxton, 2017-09-12:
https://gist.github.com/cpaxton/0d4ae182c2d3f55aa75f38308b4532be is the cprofile trace -- biggest consumers were pybullet camera matrix, loading URDF in KDL and loading it in PyBullet.

erwincoumans, 2017-09-12:
Thanks for the profile trace! Is that using DIRECT mode or GUI? Did you add this line, p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,0), before loading?

erwincoumans, 2017-09-12:
Wait, can you please create the trace using my suggested JSON method and attach the zipped json file?

p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,0) #makes loading faster
logId = p.startStateLogging(p.STATE_LOGGING_PROFILE_TIMINGS,"loadTimings.json")
#loadURDF, loadMJCF, loadSDF etc
p.stopStateLogging(logId)
p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,1) #re-enable rendering

erwincoumans, 2018-12-23:
There is a saveState / restoreState feature. Closing.

closed 2018-12-23 by erwincoumans