bulletphysics/bullet3 #1316: How to show depth and segmentation images?

ahundt, 2017-09-14:
How do I enable the segmentation and depth cameras in the GUI? Is this currently possible via something like [configureDebugVisualizer()](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.twmfawqnrl8k)?

I've searched the issues, examples, and pybullet docs, for "segmentation" and "depth", plus manually read through quite a few files with no luck. I'm able to turn on the rgb camera GUI rendering from python, but I haven't figured out how to display the segmentation and depth outside of the ExampleBrowser we've previously discussed.

I'm trying to see something like the following from [our earlier discussion](https://github.com/bulletphysics/bullet3/issues/1136#issuecomment-302985681) in my own pybullet simulation, rather than the ExampleBrowser:

[image]

erwincoumans, 2017-09-14:
This hasn't been exposed. You could use MatplotLib, or do you prefer to have the option in the GUI?

ahundt, 2017-09-14:
I'd appreciate an option in the gui! It would be helpful for ease of use and performance reasons.

ahundt, 2017-09-19:
Can details on how the segmentation mask is defined also be added to the quickstart guide? 

I don't see any information on the encoding of the segmentation mask, particularly details regarding how ids are assigned and what values will be contained in the mask. For example:

- Is there a way I can determine which joint of the arm is present, or will it only be the bodyUniqueId?
- How do I get an enumeration of all possible numerical ids that could be in the segmentation image?
- Do single object models get single ids, or does it go by parts?
- Can the granularity of ids be changed?

erwincoumans, 2017-09-19:
The segmentation mask is only the object unique id, if any, per pixel, no further details are in there.

erwincoumans, 2017-12-31:
Latest revision of getCameraImage has the option to also add the link index, next to the object unique id, for each pixel in the segmentation mask buffer. Also the quickstart guide has info about it, and example here:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/segmask_linkindex.py

Any object and link in the scene that is visible can be in the segmentation mask buffer.

closed 2017-12-31 by erwincoumans