bulletphysics/bullet3 #1321: Simulation running slow after compiling with LLVM 9.0

programming168, 2017-09-16:
Hi, 
I have being using Bulletphysics for few months. Everything was running smooth with xcode 8.3 and LLVM 8.1 .  I recently downloaded xcode 9 gm which has Apple LLVM 9.0 by default. Now stepsimulation is running very slow. 
The issue is very technical for me however i tried debugging to find out where it ran slow. I found that this particular function _maxdot_large_v0 on file LinearMath > btVector3.cpp to be running slow after compiling it with xcode 9 llvm 9.0 . Is this a issue or did i missed out something? The 3D environment contains few rigid bodies, btConvexHullShape and hinges. The app is running on ios 10.3. 
Thanks.

erwincoumans, 2017-09-16:
There is no default way to disable SIMD, except for enabling double precision mode. Safest is to edit the btScalar.h (in the __APPLE__ section remove the lines that enable SIMD/SSE. Or define BT_USE_DOUBLE_PRECISION when building Bullet and in code that includes Bullet (safest to add a #define BT_USE_DOUBLE_PRECISION at the top of btScalar.h)

You could also try manually editing btVector3::maxDot, to use the C++ code in there, instead of _maxdot_large etc.

closed 2017-09-16 by programming168

programming168, 2017-09-22:
Manually commented out btVector3::maxDot function's simd and neon codes and left out the c++ code. The c++ code is running smoothly. iOS 11 now is optimized for 64 bit so i think it could be the reason _maxdot_large_v0 is not running in full speed. 
I tried using BT_USE_DOUBLE_PRECISION but btConvexHullShape size became very small so couldn't use it. So the best option was to use c++ code on btVector3::maxDot
Thanks a lot.